esphome/feature-requests #2671: Support for WCH(Jiangsu Qin Heng) CH422G  I/O Expander

lboue, 2024-04-02:
**Describe the problem you have/What new integration you would like**

Support for WCH(Jiangsu Qin Heng) CH422G  I/O Expander

**Please describe your use case for this integration and alternatives you've tried:**

- Manufacturer WCH(Jiangsu Qin Heng)
- Mfr. Part # CH422G
- LCSC Part # C57543
- Package SOP-16
- Description 8 * 1MHz SOP-16 I/O Expanders
- Datasheet: https://www.lcsc.com/datasheet/lcsc_datasheet_2403131123_WCH-Jiangsu-Qin-Heng-CH422G_C57543.pdf

**Additional context**

lboue, 2024-05-05:
I found 2 forks from [ESP32_IO_Expander](https://github.com/esp-arduino-libs/ESP32_IO_Expander) that include drivers for CH422G I/O Expander:
 * kylix-rd: [ESP32_IO_Expander](https://github.com/kylix-rd/ESP32_IO_Expander)
 *

lboue, 2024-06-05:
[ESP32_IO_Expander Release v0.0.3](https://github.com/esp-arduino-libs/ESP32_IO_Expander/releases/tag/v0.0.3) is out 
**v0.0.3 - 2024-05-07**
Enhancements:
- Add support for CH422G from Waveshare
- Update the example to show how to use the CH422G

lboue, 2024-06-05:
> I found 2 forks from [ESP32_IO_Expander](https://github.com/esp-arduino-libs/ESP32_IO_Expander) that include drivers for CH422G I/O Expander:
> 
> * kylix-rd: [ESP32_IO_Expander](https://github.com/kylix-rd/ESP32_IO_Expander)
> * ktand: [ESP32_IO_Expander](https://github.com/ktand/ESP32_IO_Expander)
> 
> This could help to add support to ESPHome.

Or maybe it could be added a ESPHome componant like another [PCA9554 I/O Expander](https://esphome.io/components/pca9554.html).

Per4merKC, 2024-07-11:
I'm interested in this also... Thanks!

styphonthal, 2024-07-26:
Has anyone moved forward with this?

lboue, 2024-07-26:
> Has anyone moved forward with this?

No.

jesterret, 2024-08-26:
It would be great if anyone is interested in trying it out and giving me feedback on it ;)

lboue, 2024-09-01:
> It would be great if anyone is interested in trying it out and giving me feedback on it ;)

I'm in the process of testing.

lboue, 2024-09-01:
> It would be great if anyone is interested in trying it out and giving me feedback on it ;)

Working fine and tested with this config file: https://github.com/lboue/ESPHome_Waveshare_ESP32-S3_Touch_4.3/blob/main/Waveshare_ESP32-S3_Touch_4.3.yaml

lboue, 2024-09-02:
CH422G support has been merge into dev branch:
[image]

closed 2024-09-02 by lboue

ThermIQ, 2024-11-04:
I believe th proper bitmap for the CH422G controlregister shoulde be
`
    //
    // Bit:        |    7    |  6  |  5  |    4    |    3    |    2    |    1     |    0    |
    //             |   ---   | --- | --- |   ---   | ------- | ------- | -------- | ------- |
    // Value:      | [SLEEP] |  /  |  /  | [OD_EN] |         |[A_SCAN] |          | [IO_OE] |
    //             |   ---   | --- | --- |   ---   | ------- | ------- | -------- | ------- |
    // Default:    |    0    |  0  |  0  |   0     |    0    |    0    |    0     |    1    |
`

cotthemh, 2025-02-20:
there is an error when you compile it with the latest version
src/main.cpp: In function 'void setup()':
src/main.cpp:5686:57: error: invalid new-expression of abstract class type 'esphome::ch422g::CH422GGPIOPin'
 5686 |   ch422g_ch422ggpiopin_id_2 = new ch422g::CH422GGPIOPin();
      |                                                         ^
In file included from src/esphome.h:18:
src/esphome/components/ch422g/ch422g.h:49:7: note:   because the following virtual functions are pure within 'esphome::ch422g::CH422GGPIOPin':
   49 | class CH422GGPIOPin : public GPIOPin {
      |       ^~~~~~~~~~~~~
In file included from src/esphome/core/hal.h:4,
                 from src/esphome/components/adc/adc_sensor.h:6:
src/esphome/core/gpio.h:61:23: note:     'virtual esphome::gpio::Flags esphome::GPIOPin::get_flags() const'
   61 |   virtual gpio::Flags get_flags() const = 0;
      |                       ^~~~~~~~~
src/main.cpp:5692:55: error: invalid new-expression of abstract class type 'esphome::ch422g::CH422GGPIOPin'
 5692 |   ch422g_ch422ggpiopin_id = new ch422g::CH422GGPIOPin();
      |                                                       ^
/config/esphome/wallthermostat.yaml:329:57: error: invalid new-expression of abstract class type 'esphome::ch422g::CH422GGPIOPin'
  329 |       - seconds: "*"

Real-Taz, 2025-04-18:
@cotthemh 
Have you find a way around it?
I've encounter this now in version 2025.4