esphome/feature-requests #2673: Remote compilation

jrmattila, 2024-04-04:
**Describe the problem you have/What new integration you would like**

Compilation in servers with limited resources such as Raspberry Pi is slow. Compilation would be many times faster if I could _borrow_ resources from my main computer. This would ideally work in a way that Esphome web dashboard could be optionally used only as a frontend for the actual compilation that would happen in a remote server.

Basically, I would start up the compilation server with docker on my laptop. It would have an endpoint open for esphome on Raspberry Pi to connect. Once remote compilation is configured for Esphome dashboard, I would just click remote compilation and it would send the device configuration for the remote compilation server to be compiled. It would then return the firmware to the Raspberry. Or upload it directly to the esphome device assuming it has access to the same secrets and network.

**Please describe your use case for this integration and alternatives you've tried:**

Alternatively, I could compile firmwares manually on my laptop. However, I would like my configuations still to stay in the esphome-container in Raspberry so having firmwares compiled in another device based on those configurations is a bit cumbersome to achieve.

**Additional context**

This is obviously just an initial idea and there would be open questions such as whether the uploading of the firmware should happen in client or on that remote compilation server. And how would the compilations logs be transferred between servers, how to authenticate, how to deal with Esphome version mismatches between remote and client etc.

jrmattila, 2024-04-06:
One change to the Esphome-cli would make implementing this a lot easier. Ability to take the configuration from stdin would allow compilation without transferring original configuration files. You could then just pass it the whole configuration including the secrets and all the partials like as what the output of validation now returns.

This would also make it possible to implement this using 3rd party programs.

nagyrobi, 2024-04-06:
A lot of security issues would arise....

jrmattila, 2024-04-06:
> A lot of security issues would arise....

Thanks for your comment. Do you mean the stdin-part or the feature as a whole?

brotherdust, 2024-05-22:
Given how often updates come out for ESPHome, this would be very useful.

Setting something like this up would be very... non-repeatable (brittle!) unless one used a cloud service or GH Actions for the builds.

It might be better to invest the time into an automatic update feature that schedules rolling updates.