esphome/feature-requests #2674: Add support for Hynetek HUSB238 USB Power Delivery (PD) controller

lboue, 2024-04-04:
**Describe the problem you have/What new integration you would like**

Add support for Hynetek HUSB238 USB Power Delivery (PD) controller.
The [Hynetek HUSB238](https://en.hynetek.com/2421.html) is a highly integrated USB Power Delivery (PD) controller as sink role for up to 100W power rating. The HUSB238 is compatible with PD3.0 and Type-C V1.4, and it can also support Apple Divider 3, BC1.2 SDP, CDP and DCP while the source is attached.

**Please describe your use case for this integration and alternatives you've tried:**

Remote control of a circuit's supply voltage with [Adafruit USB Type C Power Delivery Dummy Breakout](https://www.adafruit.com/product/5807) boad.

**Additional context**

Arduino Library
 * [Adafruit_HUSB238](https://github.com/adafruit/Adafruit_HUSB238)

latonita, 2024-04-28:
so how do you want to use it? 
what features you want from the component?

lboue, 2024-04-28:
> so how do you want to use it? 
> 
> what features you want from the component?
> 
> 

I would set the desired PD voltage and current using i2c bus.

latonita, 2024-04-28:
> > so how do you want to use it? 
> > 
> > what features you want from the component?
> > 
> > 
> 
> I would set the desired PD voltage and current using i2c bus.

Please elaborate)
Is it one time pre pre-built in yaml or some dynamic?

Please propose config of your dream. :)

lboue, 2024-04-28:
> > > so how do you want to use it? 
> > > 
> > > what features you want from the component?
> > > 
> > > 
> > 
> > I would set the desired PD voltage and current using i2c bus.
> 
> Please elaborate)
> Is it one time pre pre-built in yaml or some dynamic?
> 
> Please propose config of your dream. :)
> 
> Any data this device produces so we need to show it as sensor? 

I'd like to use a selector to dynamically choose the voltage value.

Values can be read back for display.

lboue, 2024-05-07:
I found 2 other libraries for that chip:

- https://github.com/Duckle29/HUSB238
- https://github.com/luoluomeng/HUSB238Driver

latonita, 2024-05-07:
OK, good. I had no time yet to look deeper into this, but already manage to make base for the componenet

lboue, 2024-05-07:
> OK, good. I had no time yet to look deeper into this, but already manage to make base for the componenet

@latonita Thank you for your time. Don't hesitate to make the branch public so that I can contribute too.

latonita, 2024-05-07:
@lboue you may try what is there so far

```

external_components:
  - source: github://pr#6693
    refresh: 10s
    components:
      - husb238

husb238:
  id: husb_01

binary_sensor:
  - platform: husb238
    attached: "PD Attached"

sensor:
  - platform: husb238
    voltage: "Contracted Voltage"
    current: "Contracted Current"
    selected_voltage: "Selected Voltage"

text_sensor:
  - platform: husb238
    status: "Last request status"
    capabilities: "Capabilities"

```

lboue, 2024-05-07:
@latonita  Thank you for your time. I'll try it out straight away.

lboue, 2024-05-07:
@latonita Here are the test results for husb238 component:
I am getting this error: `Error reading HUSB238`.

```
[22:34:29]ets Jun  8 2016 00:22:57
[22:34:29]
[22:34:29]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[22:34:29]configsip: 188777542, SPIWP:0xee
[22:34:29]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[22:34:29]mode:DIO, clock div:2
[22:34:29]load:0x3fff0030,len:1184
[22:34:29]load:0x40078000,len:13132
[22:34:29]load:0x40080400,len:3036
[22:34:29]entry 0x400805e4
[22:34:29][     8][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[22:34:29][I][logger:156]: Log initialized
[22:34:29][V][esp32.preferences:059]: nvs_get_blob('233825507'): ESP_ERR_NVS_NOT_FOUND - the key might not be set yet
[22:34:29][C][ota:483]: There have been 0 suspected unsuccessful boot attempts.
[22:34:29][D][esp32.preferences:114]: Saving 1 preferences to flash...
[22:34:29][V][esp32.preferences:163]: nvs_get_blob('233825507'): ESP_ERR_NVS_NOT_FOUND - the key might not be set yet
[22:34:29][V][esp32.preferences:126]: sync: key: 233825507, len: 4
[22:34:29][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[22:34:29][I][app:029]: Running through setup()...
[22:34:29][V][app:030]: Sorting components by setup priority...
[22:34:29][I][i2c.arduino:218]: Performing I2C bus recovery
[22:34:29][    81][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=26 scl=32 freq=100000
[22:34:29][V][i2c.arduino:042]: Scanning i2c bus for active devices...
[22:34:29][C][husb238:119]: Setting up HUSB238...
[22:34:29][E][husb238:236]: Error reading HUSB238
[22:34:29][E][husb238:223]: Error reading HUSB238
[22:34:29][C][wifi:038]: Setting up WiFi...
[22:34:29][C][wifi:051]: Starting WiFi...
[22:34:29][C][wifi:052]:   Local MAC: 24:A1:60:47:99:F0
[22:34:29][V][esp32.preferences:059]: nvs_get_blob('3871777231'): ESP_ERR_NVS_NOT_FOUND - the key might not be set yet
[22:34:29][V][wifi_esp32:039]: Enabling STA.
[22:34:29][   176][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 0 - WIFI_READY
[22:34:29][V][wifi_esp32:441][arduino_events]: Event: WiFi ready
[22:34:29][   347][V][WiFiGeneric.cpp:338] _arduino_event_cb(): STA Started
[22:34:29][   348][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 2 - STA_START
[22:34:29][V][wifi_esp32:452][arduino_events]: Event: WiFi STA start
[22:34:29][E][husb238:223]: Error reading HUSB238
[22:34:29][V][sensor:043]: 'Contracted Voltage': Received new state 5.000000
[22:34:29][D][sensor:094]: 'Contracted Voltage': Sending state 5.00000 V with 0 decimals of accuracy
[22:34:29][V][sensor:043]: 'Contracted Current': Received new state 3.000000
[22:34:29][D][sensor:094]: 'Contracted Current': Sending state 3.00000 A with 2 decimals of accuracy
[22:34:29][V][sensor:043]: 'Selected Voltage': Received new state 5.000000
[22:34:29][D][sensor:094]: 'Selected Voltage': Sending state 5.00000 V with 0 decimals of accuracy
[22:34:29][V][text_sensor:013]: 'Last request status': Received new state Success
[22:34:29][D][text_sensor:064]: 'Last request status': Sending state 'Success'
[22:34:29][V][text_sensor:013]: 'Capabilities': Received new state 5V: 3.00A, 9V: 3.00A, 12V: 5.00A, 15V: 4.00A, 18V: 0.00A, 20V: 3.25A
[22:34:29][D][text_sensor:064]: 'Capabilities': Sending state '5V: 3.00A, 9V: 3.00A, 12V: 5.00A, 15V: 4.00A, 18V: 0.00A, 20V: 3.25A'
[22:34:29][W][component:237]: Component husb238 took a long time for an operation (89 ms).
[22:34:29][W][component:238]: Components should block for at most 30 ms.
[22:34:29][W][component:157]: Component wifi set Warning flag: unspecified
[22:34:29][E][husb238:223]: Error reading HUSB238
[22:34:29][E][husb238:223]: Error reading HUSB238
[22:34:29][E][husb238:223]: Error reading HUSB238
[22:34:29][E][husb238:223]: Error reading HUSB238
[22:34:29][E][husb238:223]: Error reading HUSB238
[22:34:29][E][husb238:223]: Error reading HUSB238
[22:34:29][E][husb238:223]: Error reading HUSB238
[22:34:29][E][husb238:223]: Error reading HUSB238
[22:34:29][E][husb238:223]: Error reading HUSB238
[22:34:29][E][husb238:223]: Error reading HUSB238
[22:34:29][E][husb238:223]: Error reading HUSB238
[22:34:29][E][husb238:223]: Error reading HUSB238
[22:34:29][E][husb238:223]: Error reading HUSB238
[22:34:29][E][husb238:223]: Error reading HUSB238
[22:34:29][E][husb238:223]: Error reading HUSB238
[22:34:29][E][husb238:223]: Error reading HUSB238
[22:34:29][E][husb238:223]: Error reading HUSB238
[22:34:29][E][husb238:223]: Error reading HUSB238
[22:34:29][E][husb238:223]: Error reading HUSB238
[22:34:29][E][husb238:223]: Error reading HUSB238
[22:34:29][E][husb238:223]: Error reading HUSB238
[22:34:29][E][husb238:223]: Error reading HUSB238
[22:34:29][E][husb238:223]: Error reading HUSB238
[22:34:29]b238:223]: Error reading HUSB238
```

lboue, 2024-05-25:
@jjx932 This component now works.

lboue, 2025-11-04:
- https://github.com/esphome/esphome/pull/6693

Hedda, 2026-09-24:
Also check out this ESPHome external component for the Hynetek HUSB238A (HUSB238A is different from HUSB238):

* https://github.com/sonocotta/esphome-husb238a

willpuckett, 2026-09-24:
Very excited to see this 

> Also check out this ESPHome external component for the Hynetek HUSB238A (HUSB238A is different from HUSB238):
> 
> * https://github.com/sonocotta/esphome-husb238a

closed 2026-10-12 by bdraco