esphome/feature-requests #2676: Captive Portal on demand

RFDarter, 2024-04-07:
**Describe the problem you have/What new integration you would like**
Add the possibility to start the `captive_portal` on demand from the `web_server` or even a GPIO pin.
Add the possibility do deactivate the `captive_portal` and start the `web_server` form the `captive_portal`
Make it possible to disconnect from a wifi network an start the ap with eighter the `web_server` or the `captive_portal`

**Please describe your use case for this integration and alternatives you've tried:**
Once you connected the device to a wifi network using the camptive portal there is no way to change to a different wifi network without shutting down the wifi the ESP is curently connected to, to force a start ot the ap and the captive portal.
Once connected to a wifi there is alo on way to make the device standalone anymore with using the ap insetead of connected to the wifi

**Additional context**

behnam85, 2024-06-25:
İ have a same request.

RFDarter, 2024-06-25:
> İ have a same request.

I am working on it already. I hope I can finished it soon and get it aproved and merged.

behnam85, 2024-06-26:
Thankyou @RFDarter !

grambas, 2024-09-02:
any updates on this?

rp2019433, 2024-09-15:
One more here. Totally agree with this feature. It's very useful.

behnam85, 2024-12-06:
After a few months, are we still waiting for this feature?!

RFDarter, 2025-01-30:
I finally found some time to work on it.
We need to test it on all the platforms. I only have ESP32`s here to test.

It also needs some docs to be able to get it merged. If someone would like to work on that, that would be great.

RFDarter, 2025-01-31:
https://github.com/esphome/esphome/pull/8160

iHomeegy, 2025-02-22:
What about ESP8266?