esphome/feature-requests #2680: Add SPI Support for BMP388

shingle01, 2024-04-08:
**Describe the problem you have/What new integration you would like**

As per feature request #1321 

BMP388 and BMP390 sensors supports I2C as well as SPI but ESPHome currently supports only I2C for it. It would be great if SPI support was added for this sensor as well.

**Please describe your use case for this integration and alternatives you've tried:**

Same as above

**Additional context**

Same as above

latonita, 2024-04-28:
@shingle01 do you own i2c or spi?

shingle01, 2024-04-28:
The BMP388 I have supports both.

latonita, 2024-04-28:
Okay, I will take care of this in upcoming days  if you can test

latonita, 2024-04-28:
@shingle01 
try new PR
```
esphome:
  name: bmp3xx-test

esp32:
  board: esp32dev
  framework:
    type: arduino

# esp8266:
#   board: nodemcuv2

logger:
  level: DEBUG

external_components:
  - source: github://pr#6652
    refresh: 10s
    components: [bmp3xx_base, bmp3xx_i2c, bmp3xx_spi]

spi:
  clk_pin: 19
  mosi_pin: 20
  miso_pin: 21

i2c:
  sda: GPIO14
  scl: GPIO26
  scan: true
  frequency: 5kHz

sensor:
  - platform: bmp3xx_spi
    cs_pin: 13
    temperature:
      name: "Outside Temperature"
      oversampling: 16x
    pressure:
      name: "Outside Pressure"
    update_interval: 60s
  - platform: bmp3xx_i2c
    address: 0x77
    temperature:
      name: "Outside Temperature"
      oversampling: 16x
    pressure:
      name: "Outside Pressure"
    update_interval: 60s
```

closed 2024-05-02 by jesserockz

shingle01, 2024-05-11:
Sorry for the delay.  Testing SPI only I get an error:

`[13:30:09][C][bmp3xx.sensor:151]: BMP3XX:
**[13:30:09][C][bmp3xx.sensor:152]:   Type: Unknown Chip Type (0xFF)**
**[13:30:09][E][bmp3xx.sensor:163]: BMP3XX has wrong chip ID (reported id: 0xFF) - please check if you are really using a BMP 388 or BMP 390**
[13:30:09][C][bmp3xx.sensor:172]:   IIR Filter: OFF
[13:30:09][C][bmp3xx.sensor:173]:   Update Interval: 60.0s
[13:30:09][C][bmp3xx.sensor:175]:   Temperature 'Outside Temperature'
[13:30:09][C][bmp3xx.sensor:175]:     Device Class: 'temperature'
[13:30:09][C][bmp3xx.sensor:175]:     State Class: 'measurement'
[13:30:09][C][bmp3xx.sensor:175]:     Unit of Measurement: '°C'
[13:30:09][C][bmp3xx.sensor:175]:     Accuracy Decimals: 1
[13:30:09][C][bmp3xx.sensor:176]:     Oversampling: 16x
[13:30:09][C][bmp3xx.sensor:179]:   Pressure 'Outside Pressure'
[13:30:09][C][bmp3xx.sensor:179]:     Device Class: 'pressure'
[13:30:09][C][bmp3xx.sensor:179]:     State Class: 'measurement'
[13:30:09][C][bmp3xx.sensor:179]:     Unit of Measurement: 'hPa'
[13:30:09][C][bmp3xx.sensor:179]:     Accuracy Decimals: 1
[13:30:09][C][bmp3xx.sensor:180]:     Oversampling: 16x
**[13:30:09][E][component:082]:   Component bmp3xx_base is marked FAILED`**

latonita, 2024-05-11:
@shingle01 please test new PR for SPI
  - source: github://pr#6722

shingle01, 2024-05-11:
Better but getting erroneous results

[08:04:23][D][bmp3xx.sensor:212]: Got temperature=-98.8°C
pressure=-2243.0hPa
[08:04:23][D][sensor:094]: 'Outside Temperature': Sending state -98.76825
°C with 1 decimals of accuracy
[08:04:23][D][sensor:094]: 'Outside Pressure': Sending state -2242.96802
hPa with 1 decimals of accuracy

latonita, 2024-05-11:
Please enable very_verbose logging and attach the log.

shingle01, 2024-05-11:
[08:47:15][C][bmp3xx.sensor:151]: BMP3XX: [08:47:15][C][bmp3xx.sensor:152]:
Type: BMP 388 (0x50) [08:47:15][C][bmp3xx.sensor:172]: IIR Filter: OFF
[08:47:15][C][bmp3xx.sensor:173]: Update Interval: 60.0s
[08:47:15][C][bmp3xx.sensor:175]:
Temperature 'Outside Temperature' [08:47:15][C][bmp3xx.sensor:175]: Device
Class: 'temperature' [08:47:15][C][bmp3xx.sensor:175]: State Class:
'measurement' [08:47:15][C][bmp3xx.sensor:175]: Unit of Measurement: '°C'
[08:47:15][C][bmp3xx.sensor:175]: Accuracy Decimals: 1
[08:47:15][C][bmp3xx.sensor:176]:
Oversampling: 16x [08:47:15][C][bmp3xx.sensor:179]: Pressure 'Outside
Pressure' [08:47:15][C][bmp3xx.sensor:179]: Device Class: 'pressure'
[08:47:15][C][bmp3xx.sensor:179]: State Class: 'measurement'
[08:47:15][C][bmp3xx.sensor:179]:
Unit of Measurement: 'hPa' [08:47:15][C][bmp3xx.sensor:179]: Accuracy
Decimals: 1 [08:47:15][C][bmp3xx.sensor:180]: Oversampling: 16x
[08:47:15][C][captive_portal:088]:
Captive Portal: [08:47:15][C][mdns:115]: mDNS: [08:47:15][C][mdns:116]:
Hostname: sensor-box-2 [08:47:15][V][mdns:117]: Services:
[08:47:15][V][mdns:119]:
- _esphomelib, _tcp, 6053 [08:47:15][V][mdns:121]: TXT: version =
2024.6.0-dev [08:47:15][V][mdns:121]: TXT: mac = 0cb815f51084
[08:47:15][V][mdns:121]:
TXT: platform = ESP32 [08:47:15][V][mdns:121]: TXT: board = esp32dev
[08:47:15][V][mdns:121]: TXT: network = wifi [08:47:15][V][mdns:121]: TXT:
api_encryption = Noise_NNpsk0_25519_ChaChaPoly_SHA256 [08:47:15][C][ota:096]:
Over-The-Air Updates: [08:47:15][C][ota:097]: Address: 192.168.86.118:3232
[08:47:15][C][ota:100]: Using Password. [08:47:15][C][ota:103]: OTA
version: 2. [08:47:15][C][api:139]: API Server: [08:47:15][C][api:140]:
Address: 192.168.86.118:6053 [08:47:15][C][api:142]: Using noise
encryption: YES [08:47:35][VV][scheduler:226]: Running interval '' with
interval=60000 last_execution=4294939105 (now=31810)
[08:47:35][VV][scheduler:226]:
Running interval 'update' with interval=60000 last_execution=4294939135
(now=31842) [08:47:35][V][bmp3xx.sensor:189]: Sending conversion
request... [08:47:35][VV][bmp3xx.sensor:202]:
measurement time 67 [08:47:35][VV][scheduler:032]: set_timeout(name='data',
timeout=67) [08:47:35][VV][scheduler:226]: Running timeout 'data' with
interval=67 last_execution=31863 (now=31930) [08:47:35][VV][bmp3xx.sensor:349]:
data ready status 8 [08:47:35][D][bmp3xx.sensor:212]: Got
temperature=-872.0°C pressure=-23136.3hPa [08:47:35][V][sensor:043]:
'Outside Temperature': Received new state -871.951050
[08:47:35][D][sensor:094]:
'Outside Temperature': Sending state -871.95105 °C with 1 decimals of
accuracy [08:47:35][VV][api.service:140]: send_sensor_state_response:
SensorStateResponse { key: 2259435129 state: -871.951 missing_state: NO }
[08:47:35][V][sensor:043]: 'Outside Pressure': Received new state
-23136.251953 [08:47:35][D][sensor:094]: 'Outside Pressure': Sending state
-23136.25195 hPa with 1 decimals of accuracy [08:47:35][VV][api.service:140]:
send_sensor_state_response: SensorStateResponse { key: 3243141300 state:
-23136.3 missing_state: NO } [08:47:35][W][component:237]: Component
bmp3xx_base took a long time for an operation (79 ms).
[08:47:35][W][component:238]:
Components should block for at most 30 ms. [08:47:55][VV][api.service:654]:
on_ping_request: PingRequest {} [08:47:55][VV][api.service:043]:
send_ping_response: PingResponse {} [08:48:12][VV][api.service:654]:
on_ping_request: PingRequest {} [08:48:12][VV][api.service:043]:
send_ping_response: PingResponse {} [08:48:35][VV][scheduler:226]: Running
interval '' with interval=60000 last_execution=31809 (now=91810)
[08:48:35][VV][scheduler:226]:
Running interval 'update' with interval=60000 last_execution=31839
(now=91841) [08:48:35][V][bmp3xx.sensor:189]: Sending conversion
request... [08:48:35][VV][bmp3xx.sensor:202]:
measurement time 67 [08:48:35][VV][scheduler:032]: set_timeout(name='data',
timeout=67) [08:48:35][VV][scheduler:226]: Running timeout 'data' with
interval=67 last_execution=91862 (now=91929) [08:48:35][VV][bmp3xx.sensor:349]:
data ready status 8 [08:48:35][D][bmp3xx.sensor:212]: Got
temperature=69.2°C pressure=-1478.2hPa [08:48:35][V][sensor:043]: 'Outside
Temperature': Received new state 69.150589 [08:48:35][D][sensor:094]:
'Outside Temperature': Sending state 69.15059 °C with 1 decimals of
accuracy [08:48:35][VV][api.service:140]:
send_sensor_state_response: SensorStateResponse { key: 2259435129 state:
69.1506 missing_state: NO } [08:48:35][V][sensor:043]: 'Outside Pressure':
Received new state -1478.219727 [08:48:35][D][sensor:094]: 'Outside
Pressure': Sending state -1478.21973 hPa with 1 decimals of accuracy
[08:48:35][VV][api.service:140]:
send_sensor_state_response: SensorStateResponse { key: 3243141300 state:
-1478.22 missing_state: NO } [08:48:35][W][component:237]: Component
bmp3xx_base took a long time for an operation (79 ms).
[08:48:35][W][component:238]:
Components should block for at most 30 ms.

latonita, 2024-05-12:
@shingle01 apologies, its hard to read the log, it is deformed
would you please paste it in via ``` signs in the github ui, not via email. thanks

shingle01, 2024-05-12:
```
[09:40:41][C][bmp3xx.sensor:151]: BMP3XX:
[09:40:41][C][bmp3xx.sensor:152]:   Type: BMP 388 (0x50)
[09:40:41][C][bmp3xx.sensor:172]:   IIR Filter: OFF
[09:40:41][C][bmp3xx.sensor:173]:   Update Interval: 60.0s
[09:40:41][C][bmp3xx.sensor:175]:   Temperature 'Outside Temperature'
[09:40:41][C][bmp3xx.sensor:175]:     Device Class: 'temperature'
[09:40:41][C][bmp3xx.sensor:175]:     State Class: 'measurement'
[09:40:41][C][bmp3xx.sensor:175]:     Unit of Measurement: '°C'
[09:40:41][C][bmp3xx.sensor:175]:     Accuracy Decimals: 1
[09:40:41][C][bmp3xx.sensor:176]:     Oversampling: 16x
[09:40:41][C][bmp3xx.sensor:179]:   Pressure 'Outside Pressure'
[09:40:41][C][bmp3xx.sensor:179]:     Device Class: 'pressure'
[09:40:41][C][bmp3xx.sensor:179]:     State Class: 'measurement'
[09:40:41][C][bmp3xx.sensor:179]:     Unit of Measurement: 'hPa'
[09:40:41][C][bmp3xx.sensor:179]:     Accuracy Decimals: 1
[09:40:41][C][bmp3xx.sensor:180]:     Oversampling: 16x
```

```
[09:41:16][VV][scheduler:226]: Running interval 'update' with interval=60000 last_execution=4294952051 (now=44761)
[09:41:16][V][bmp3xx.sensor:189]: Sending conversion request...
[09:41:16][VV][bmp3xx.sensor:202]: measurement time 67
[09:41:16][VV][scheduler:032]: set_timeout(name='data', timeout=67)
[09:41:16][VV][scheduler:226]: Running timeout 'data' with interval=67 last_execution=44776 (now=44844)
[09:41:16][VV][bmp3xx.sensor:349]: data ready status 8
[09:41:16][D][bmp3xx.sensor:212]: Got temperature=-849.3°C pressure=-19404.6hPa
[09:41:16][V][sensor:043]: 'Outside Temperature': Received new state -849.316772
[09:41:16][D][sensor:094]: 'Outside Temperature': Sending state -849.31677 °C with 1 decimals of accuracy
[09:41:16][VV][api.service:140]: send_sensor_state_response: SensorStateResponse {
  key: 2259435129
  state: -849.317
  missing_state: NO
}
[09:41:16][V][sensor:043]: 'Outside Pressure': Received new state -19404.640625
[09:41:16][D][sensor:094]: 'Outside Pressure': Sending state -19404.64062 hPa with 1 decimals of accuracy
[09:41:16][VV][api.service:140]: send_sensor_state_response: SensorStateResponse {
  key: 3243141300
  state: -19404.6
  missing_state: NO
}
[09:41:16][W][component:237]: Component bmp3xx_base took a long time for an operation (89 ms).
[09:41:16][W][component:238]: Components should block for at most 30 ms.
```

latonita, 2024-06-05:
@shingle01 can you please test one more time with very verbose logging - i would need few readings done.
there is a chance first byte coming from the device over SPI is always 0xFF and shall be skipped. need to check that idea.
thanks

jrydval, 2024-07-27:
Good afternoon!

Yes, it seems that he first byte is all the time 0xFF. 

The log is attached:

```
[14:41:17][VV][scheduler:225]: Running interval 'update' with interval=60000 last_execution=4294960762 (now=53466)
[14:41:17][V][bmp3xx.sensor:189]: Sending conversion request...
[14:41:17][VV][spi-esp-arduino:021]: 8266 mangled SPI mode 0x0
[14:41:17][VV][bmp3xx_spi.sensor:045]: Write register 1b (sent 1b), Byte received: 13
[14:41:17][VV][bmp3xx.sensor:202]: measurement time 67
[14:41:17][VV][scheduler:032]: set_timeout(name='data', timeout=67)
[14:41:17][VV][scheduler:225]: Running timeout 'data' with interval=67 last_execution=53491 (now=53558)
[14:41:17][VV][spi-esp-arduino:021]: 8266 mangled SPI mode 0x0
[14:41:17][VV][bmp3xx_spi.sensor:037]: Read register 11 (sent 91), Byte received: 08
[14:41:17][VV][bmp3xx.sensor:349]: data ready status 8
[14:41:17][VV][spi-esp-arduino:021]: 8266 mangled SPI mode 0x0
[14:41:17][VV][bmp3xx_spi.sensor:057]: Read bytes. register 04 (sent 84), Bytes received: FF.40.AC.70.00.66 (6)
[14:41:17][D][bmp3xx.sensor:212]: Got temperature=-57.5°C pressure=674.0hPa
[14:41:17][V][sensor:043]: 'Lolin1 Temperature': Received new state -57.477680
[14:41:17][D][sensor:093]: 'Lolin1 Temperature': Sending state -57.47768 °C with 1 decimals of accuracy
[14:41:17][V][sensor:043]: 'Lolin1 Pressure': Received new state 674.040649
[14:41:17][VV][sensor.filter:014]: Filter(0x3fff50c4)::input(674.040649)
[14:41:17][VV][sensor.filter:021]: Filter(0x3fff50c4)::output(674.040649) -> SENSOR
[14:41:17][D][sensor:093]: 'Lolin1 Pressure': Sending state 674.04065 hPa with 1 decimals of accuracy
[14:41:17][VV][spi-esp-arduino:021]: 8266 mangled SPI mode 0x0
[14:41:17][VV][bmp3xx_spi.sensor:045]: Write register 1b (sent 1b), Byte received: 03
[14:41:17][W][component:237]: Component bmp3xx_base took a long time for an operation (96 ms).
[14:41:17][W][component:238]: Components should block for at most 30 ms.
[14:41:27][VV][scheduler:225]: Running interval '' with interval=60000 last_execution=3345 (now=63345)
[14:41:43][VV][api.connection:132]: Sending keepalive PING...
[14:41:43][VV][api.service:037]: send_ping_request: PingRequest {}
[14:41:43][VV][api.service:663]: on_ping_response: PingResponse {}
[14:42:17][VV][scheduler:225]: Running interval 'update' with interval=60000 last_execution=53466 (now=113471)
[14:42:17][V][bmp3xx.sensor:189]: Sending conversion request...
[14:42:17][VV][spi-esp-arduino:021]: 8266 mangled SPI mode 0x0
[14:42:17][VV][bmp3xx_spi.sensor:045]: Write register 1b (sent 1b), Byte received: 13
[14:42:17][VV][bmp3xx.sensor:202]: measurement time 67
[14:42:17][VV][scheduler:032]: set_timeout(name='data', timeout=67)
[14:42:17][VV][scheduler:225]: Running timeout 'data' with interval=67 last_execution=113495 (now=113562)
[14:42:17][VV][spi-esp-arduino:021]: 8266 mangled SPI mode 0x0
[14:42:17][VV][bmp3xx_spi.sensor:037]: Read register 11 (sent 91), Byte received: 08
[14:42:17][VV][bmp3xx.sensor:349]: data ready status 8
[14:42:17][VV][spi-esp-arduino:021]: 8266 mangled SPI mode 0x0
[14:42:17][VV][bmp3xx_spi.sensor:057]: Read bytes. register 04 (sent 84), Bytes received: FF.50.A5.70.80.56 (6)
[14:42:17][D][bmp3xx.sensor:212]: Got temperature=-91.0°C pressure=468.3hPa
[14:42:17][V][sensor:043]: 'Lolin1 Temperature': Received new state -91.014580
[14:42:17][D][sensor:093]: 'Lolin1 Temperature': Sending state -91.01458 °C with 1 decimals of accuracy
[14:42:17][V][sensor:043]: 'Lolin1 Pressure': Received new state 468.266296
[14:42:17][VV][sensor.filter:014]: Filter(0x3fff50c4)::input(468.266296)
[14:42:17][VV][sensor.filter:021]: Filter(0x3fff50c4)::output(468.266296) -> SENSOR
[14:42:17][D][sensor:093]: 'Lolin1 Pressure': Sending state 468.26630 hPa with 1 decimals of accuracy
[14:42:17][VV][spi-esp-arduino:021]: 8266 mangled SPI mode 0x0
[14:42:17][VV][bmp3xx_spi.sensor:045]: Write register 1b (sent 1b), Byte received: 03
[14:42:17][W][component:237]: Component bmp3xx_base took a long time for an operation (97 ms).
[14:42:17][W][component:238]: Components should block for at most 30 ms.
[14:42:27][VV][scheduler:225]: Running interval '' with interval=60000 last_execution=63345 (now=123345)
[14:42:43][VV][api.connection:132]: Sending keepalive PING...
[14:42:43][VV][api.service:037]: send_ping_request: PingRequest {}
[14:42:43][VV][api.service:663]: on_ping_response: PingResponse {}
[14:43:17][VV][scheduler:225]: Running interval 'update' with interval=60000 last_execution=113466 (now=173466)
[14:43:17][V][bmp3xx.sensor:189]: Sending conversion request...
[14:43:17][VV][spi-esp-arduino:021]: 8266 mangled SPI mode 0x0
[14:43:17][VV][bmp3xx_spi.sensor:045]: Write register 1b (sent 1b), Byte received: 13
[14:43:17][VV][bmp3xx.sensor:202]: measurement time 67
[14:43:17][VV][scheduler:032]: set_timeout(name='data', timeout=67)
[14:43:17][VV][scheduler:225]: Running timeout 'data' with interval=67 last_execution=173491 (now=173559)
[14:43:17][VV][spi-esp-arduino:021]: 8266 mangled SPI mode 0x0
[14:43:17][VV][bmp3xx_spi.sensor:037]: Read register 11 (sent 91), Byte received: 08
[... 25 lines cut ...]
[14:44:17][VV][spi-esp-arduino:021]: 8266 mangled SPI mode 0x0
[14:44:17][VV][bmp3xx_spi.sensor:037]: Read register 11 (sent 91), Byte received: 08
[14:44:17][VV][bmp3xx.sensor:349]: data ready status 8
[14:44:17][VV][spi-esp-arduino:021]: 8266 mangled SPI mode 0x0
[14:44:17][VV][bmp3xx_spi.sensor:057]: Read bytes. register 04 (sent 84), Bytes received: FF.70.9E.70.70.45 (6)
[... cut ...]
[14:44:17][W][component:237]: Component bmp3xx_base took a long time for an operation (97 ms).
[14:44:17][W][component:238]: Components should block for at most 30 ms.
[14:44:27][VV][scheduler:225]: Running interval '' with interval=60000 last_execution=183345 (now=243345)
[14:44:43][VV][api.connection:132]: Sending keepalive PING...
[14:44:43][VV][api.service:037]: send_ping_request: PingRequest {}
[14:44:43][VV][api.service:663]: on_ping_response: PingResponse {}
```

Have a sunny weekend!

Jan