esphome/feature-requests #2682: Option to pre-build firmware updates in home assistant add-on

pzich, 2024-04-11:
**Describe the problem you have/What new integration you would like**

The ESPHome Home Assistant add-on notifies you that your device firmware is ready to update, and lets you perform the update from the app. Unfortunately in my case, it takes a _long_ time to build the new firmware on my underpowered Home Assistant VM. I like to make sure that the device is still functioning as expected after doing an update, so it would be great to minimize the time between pressing update and the device being update.

Ideally, I could enable a configuration option in the add-on that, when it detects my devices have an update, automatically starts building them. Then when I click update/install, it only has to spend a few seconds uploading the firmware and restarting the device.

**Please describe your use case for this integration and alternatives you've tried:**

As an alternative, I've considered writing a script on my computer that attempts to do effectively the same checking and compiling as the add-on does, but this is a lot of duplicate work and requires me to go to my computer to do the install instead of the home assistant app.

**Additional context**

BertVereecke, 2024-06-27:
+1
In my case, the compilation takes 2'53". The upload and installation completes in 16 seconds.

PhilippeP62, 2024-08-01:
Yes , updating my onju with oww takes less than 2 minutes , more than 10 with mww ... and I have 4 of them , so prebuilding would speed things a lot.

DaVinylSmith, 2024-08-02:
+1 here.

once of my devices is a home power monitor. It would help greatly to be able to build the firmware ahead of time then schedule the actual flashing at a point of least impact (as some values get reset on device reboots)