esphome/feature-requests #2683: Option to set default update method

pzich, 2024-04-11:
**Describe the problem you have/What new integration you would like**

The ESPHome dashboard for the Home Assistant add-on shows all of my devices and if they need updates. When I click the Update button, I'm presented with a list of options for how I want to perform the update.

[image]

In my case, I _always_ want to update wirelessly. It would be great if a setting could be added where I can set the default update method. This option would the be used when clicking the Update button, and the other options could be access one-off either by using a modifier key like shift-click, or with an item in the `[ ⋮ ]` menu.

**Please describe your use case for this integration and alternatives you've tried:**

Installing the update from Home Assistant's Settings page seems to default to wireless updating, but there's no way to see the output of the compilation and upload logs when installing from here. I like to monitor this to make sure everything works successfully.

**Additional context**

bruce3322, 2024-07-16:
+1 for this

WagnerSystems, 2024-08-09:
+1 from me also