esphome/feature-requests #2684: new Sensor Voltage ZMPT101B

gaitolini, 2024-04-12:
**Describe the problem you have/What new integration you would like**
I suggest implementing a new voltage sensor for the AC Voltage Sensor 0 to 250V Voltmeter ZMPT101B device, ideal for home automation projects using Arduino or other microcontrolled platforms.

**Please describe your use case for this integration and alternatives you've tried:**
The AC Voltage Sensor 0 to 250V Voltmeter ZMPT101B is a high-precision module designed to detect the presence of alternating voltage in a circuit or to measure the voltage value. For those developing home automation projects, the AC Voltage Sensor 0 to 250V Voltmeter ZMPT101B is extremely important. It can indicate whether a lamp is on or off, whether a motor is on or off, regardless of whether the circuit is controlled by an application/web page or by a switch/button. Additionally, this sensor allows the implementation of a project aimed at monitoring voltage values in the alternating current network and can be used as a voltmeter.

My idea is to create a sensor for use on ESP32 and ESP8266 devices.

[image]

**Additional context**
# Example configuration entry
```yaml
sensor:
    - platform: zmpt101b
      pin: 34
      name: "Measured Voltage"
      frequency: 50hz
      num_of_samples: 50
      calibration: 86
      phase_shift: 1.7

or using the ADC sensor:

  - platform: zmpt_sensor
    sensor: adc01_sensor39
    name: zmpt
    update_interval: 10s
    filters:
      - calibrate_linear:
          - 0 -> 0
          - 0.1333 -> 4.0
    unit_of_measurement: V

  - platform: adc
    pin: GPIO39
    id: adc01_sensor39

rafal83, 2025-03-07:
Hi,

You can use my sensor => https://github.com/rafal83/zmpt101b

rafal83, 2025-07-24:
> I also think this sensor would be very usefull.

Try this : https://github.com/rafal83/zmpt101b