esphome/feature-requests #2685: ILI9488 alternative color modes

rubdos, 2024-04-14:
**Describe the problem you have/What new integration you would like**
I have an ILI9844, but the refresh time is quite long in 40MHz mode, and the 80MHz mode seems to crash my poorly designed PCB.

**Please describe your use case for this integration and alternatives you've tried:**
 It could be beneficial to work in reduced color modes.

**Additional context**

Display color modes:
* Full color modes:
  * 16.7M colors with dithering function (24-bit data, R: 8-bit, G: 8-bit, B: 8-bit)
  * 262K colors (18-bit data, R: 6-bit, G: 6-bit, B: 6-bit)
* Reduced color modes:
  * 65K colors (16-bit data, R: 5-bit, G: 6-bit, B: 5-bit)
  * 8 colors (3-bit data, R: 1-bit, G: 1-bit, B: 1-bit)

#1125 and #1715 seem to request initial support for this display, but it seems to work fine already in current ESPhome :-)

nielsnl68, 2024-04-14:
i'm working on a new design for the ili9xxx driver with more controle on how to talk to the display,
The first step is to allow different IO interfaces above the current SPI interface, like Parallel 8 and 16 but mode and esp32-S3's RGB mode.
Next step will be more control over the color depths and the way it pixels are send to the display (little endian, byte aligned and color order) 

Sadly i do not have much time to work on it but it is slowly coming to live.

rubdos, 2024-04-14:
That is *really* cool to hear, thank you very much! It's already really lovely to play with.

For your inspiration and motivation, this is what I built with the current state:

[image]

[image]

It's an ESP32-S3 with a bunch of connectors to control and measure various parts of my aquarium.