esphome/feature-requests #2686: MAC filter for bluetooth_proxy

g-kiss, 2024-04-14:
**Describe the problem you have/What new integration you would like**

It would be very useful if optionally I could specify the MAC addresses that bluetooth_proxy have to handle. That way, it would not send a lot of unnecessary data over the Wi-Fi network.
Like this:
```
bluetooth_proxy:
  active: false
  mac_address:
    - 11:11:11:11:11:11
    - 22:22:22:22:22:22
```

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

CReimer, 2024-05-02:
+1 from me. Wanted to ask for this myself
In my case I have a few Apple devices. (Maybe Apple TVs) which push out an insane amount of data over BLE. I want that data ignored as I suspect that much data breaks WiFi on the ESP32

I'd like to extend @g-kiss request with the possibility to add a partial mac address to the whitelist e.g. ```A4:C1:38:*```

KiAchim, 2026-10-09:
[esphome-web-583d2c-logs(5).txt](https://github.com/user-attachments/files/33249217/esphome-web-583d2c-logs.5.txt)
This from my point of view is a must have. My situation is: I have 8 BLE-Devices located in 6 rooms. 5 ESP devices should proxy with them, each with one or two BLEs, and establish GATT protocol. It is impossible to make sure the individual ESPs only speak to **_their_** one or two devices. So there are a tremendous activities on each ESP trying to build up communication to BLE-Devices further away, tearing down the communication, blocking GATT-slots preventing the establishing of communication to  **_their_** devices.
in the log attached you see that there is traffic to 00:1A:22:06:F2:49, 00:1A:22:0F:06:40, but the connection to the two devices 00:1A:22:06:F1:CB and 00:1A:22:06:F1:CB, which are located close to the ESP, is not established.

closed 2026-11-14 by g-kiss