esphome/feature-requests #2687: Add support for climate component, platform: modbus

ionutgogu, 2024-04-15:
**Describe the problem you have/What new integration you would like**

Possibility to create a climate component using platform: modbus to be able to write/read directly to registry addresses or to an already defined sensor: 

Example: 

- name: "hp_water_climate"
      address: 1020
      register_type: holding
      value_type: S_WORD
      multiply: 10
      max_temp: 60
      min_temp: 15
      offset: 0
      precision: 1
      scale: 0.5
      temp_step: 0.5
      temperature_unit: C
      operation_modes:
          - "on"
           - "cool"
           - "heat"

**Please describe your use case for this integration and alternatives you've tried:**

Integrate a heatpump into homeassistant using modbus rtu and esphome

**Additional context**

ssieb, 2024-04-15:
Does the modbus device actually have climate controls?  Do you have documentation?

ionutgogu, 2024-04-15:
Hi,
unfortunately I did not saw your question on discord, it is not related to one  device, almost all hvac devices with modbus have climate option, see homeassiatant core modbus integration:

https://www.home-assistant.io/integrations/modbus/

Now I am using an USB_RS485 adapter, the above integration and will be nice to use ESPHome.