esphome/feature-requests #2689: Make `esp32_ble_tracker` and `esp32_ble_beacon` interoperable

agittins, 2024-04-16:
**Describe the problem you have/What new integration you would like**

I would like to be able to configure my esphome `bluetooth_proxy`'s to also act as `esp32_ble_beacon`s. Currently, adding a `esp32_ble_beacon:` section to the yaml generates an error stating that `Component esp32_ble_beacon cannot be used together with component esp32_ble_tracker.`

**Please describe your use case for this integration and alternatives you've tried:**

I created the [Bermuda](/agittins/bermuda) custom integration which uses gathered advertisements to create location information in HA for BLE beacons, including room-based localisation.

A major goal for the project is full trilateration of the bluetooth environment, which in part requires solving the distances between esphome proxies. I have ways of doing this, but accuracy would benefit from being able to configure the proxies to periodically send out  a beacon packet so that the other proxies can estimate their distance from it. A side-effect of this is the requirement that proxies do this at different times, so retaining the functionality of the `min_interval` and `max_interval` options of `esp32_ble_beacon` would be necessary, and a working config would probably require increasing them significantly so as to not adversely affect `bluetooth_proxy` performance.

Ideally I would probably want to configure them to only send out adverts every hour or so, since the integration will maintain a record of estimated distances.

**Additional context**

Key features desired:
- Allow `esp32_ble_tracker`, `bluetooth_proxy` and `esp32_ble_beacon` to co-exist
- Ideally, allow the beacon to be turned on and off via lambda or have long interval times specified for better proxy performance

The Shelly Plus (or Gen2?) devices appear to do this according to user reports, I have no idea if there's any shared lineage there.

This pull request https://github.com/esphome/esphome/pull/3474 discusses the feature, and looked to have nearly got there but for a change in direction re implementation, but I have not been able to find subsequent attempts at the feature. My understanding from that is that implementing a core `esp32_ble` component would be the primary goal, then making the other ble-related components dependent on that.

Assumptions/observations tested on esphome 2024.3.2, using `esp-idf` framework, `wemos_d1_mini32` board. Happy to provide attempted yaml if relevant.

jaymunro, 2024-04-16:
I can confirm my Shelly devices are able to see each other's RSSI and distances between them can be estimated as well.  

@jesserockz's post [here](https://github.com/esphome/esphome/pull/3474#issuecomment-1216080330) seems to indicate that the esp32_ble component exists and it may not be a breaking change to migrate the tracker and beacon components to use it so they can co-exist.

jesserockz, 2024-04-18:
Correct, the ble_beacon component needs to be updated to inherit from esp32_ble then it can co-exist

closed 2024-07-22 by jesserockz