esphome/feature-requests #2692: Please add [Download firmware] if OTA goes wrong

QuAzI, 2024-04-17:
**Describe the problem you have/What new integration you would like**
Please add [Download firmware] for manual download when OTA failed

**Please describe your use case for this integration and alternatives you've tried:**
Sometimes previous configuration get into non-bootable device or device reboots in a loop. So, new firmware can't be uploaded

**Additional context**
[image]

MiguelAngelLV, 2024-04-20:
You can download the firmware from previous menu, when you select Wireless o USB, you can select "Manually"  and it download the firmware

QuAzI, 2024-04-20:
In this way, the build process starts again, and the user has to wait while it finishes

HenryLoenwind, 2024-05-15:
It probably would make more sense to add some brains to the "install" menu, so it notices it already has an up-to-date firmware file and doesn't need to compile a new one.