esphome/feature-requests #2693: Support ds2483-800

alainstark, 2024-04-18:
**Hello, I have write code to support ds2482-800 to the component ds248x**

**I have comment this feature at the end of this [feature-request](https://github.com/esphome/feature-requests/issues/621)**

**I am not a git expert, can you help me to add the ds248x component to esphome ?**

nagyrobi, 2024-04-18:
Post a link

alainstark, 2024-04-18:
[ds248x.zip](https://github.com/esphome/feature-requests/files/15030038/ds248x.zip)

This is the code. Il that OK ?

alainstark, 2024-04-19:
Excuse me but I'm really bad at git. I closed this request by mistake.
I don't know how to go back. Should I open a new request?

nagyrobi, 2024-04-19:
@alainstark We need a doc description too, could you make it?

alainstark, 2024-04-19:
Yes, is this [link](https://esphome.io/guides/contributing#contributing-to-esphome-docs) the right thing to learn ?

alainstark, 2024-04-21:
Have you incrase the verbosity of the logs ?

alainstark, 2024-04-21:
I belive that you should not use id: ds2482_800_ch1_1. This syntax is for multiple ds248x on the same I²C bus. Comment this line and text again.

alainstark, 2024-04-21:
And this is the doc :-)
[ds248x-doc.zip](https://github.com/esphome/feature-requests/files/15052930/ds248x-doc.zip)

csabter, 2024-04-21:
Thank you very much!

It's work! :)

I don't know why, but there were some indentation problem in the sensor.py file (i unziped and opened it from Linux, Geany and Kate):

After unzip:
[...]
async def to_code(config):
    hub = await cg.get_variable(config[CONF_DALLAS_ID])
    var = await sensor.new_sensor(config)
    
    if CONF_ADDRESS in config:
        cg.add(var.set_address(config[CONF_ADDRESS]))
    else:
        cg.add(var.set_index(config[CONF_INDEX]))
        
        if CONF_RESOLUTION in config:
            cg.add(var.set_resolution(config[CONF_RESOLUTION]))
            
            if CONF_CHANNEL in config:
                cg.add(var.set_channel(config[CONF_CHANNEL]))
                
                cg.add(var.set_parent(hub))
                
                cg.add(hub.register_sensor(var))

After orrection:
[...]
async def to_code(config):
    hub = await cg.get_variable(config[CONF_DALLAS_ID])
    var = await sensor.new_sensor(config)
    
    if CONF_ADDRESS in config:
        cg.add(var.set_address(config[CONF_ADDRESS]))
    else:
        cg.add(var.set_index(config[CONF_INDEX]))
        
    if CONF_RESOLUTION in config:
        cg.add(var.set_resolution(config[CONF_RESOLUTION]))
        
    if CONF_CHANNEL in config:
        cg.add(var.set_channel(config[CONF_CHANNEL]))
            
    cg.add(var.set_parent(hub))
            
    cg.add(hub.register_sensor(var))

I will change the id sytax.  

I am very gateful for your help and this code and documentation!

(Sorry, my English is not very good...)

alainstark, 2024-04-22:
Yes it's my fault. Before generating the zip archive, I opened the code in emacs and mechanically indented the code without testing.
I'm going to try to train myself in git to set up this new component in esphome. If anyone wants to help me, I'm all ears. Concerning English, I myself (I am French) am not an expert and I admit to using Google translate.

nagyrobi, 2024-04-22:
Less pro, but can be done even from web UI of GitHub.
Clone ESPHome repo to your account.
Create a new branch from DEV.
Create a new subdirectory in the components for your component. Upload the files.
Contribute back by comparing changes against dev branches, and making a pull request.
Similar for doc.

csabter, 2024-04-22:
I'd like to help, but unfortunately i don't know git at all (basically, i work with hardware), i hope you can somehow get it together with ESPHome Repo! (I'll write as soon as i find someone who could help.) I think it absolutely fills a gap!
Thank you!

alainstark, 2024-04-22:
Ok nagyrobi. I used the non pro method.
Is that correct ?
[esphome](https://github.com/alainstark/esphome)
[esphome_doc](https://github.com/alainstark/esphome-docs)
If yes, what must I do yet ?

nagyrobi, 2024-04-23:
Now you see in your esphome repo, at the top of the page a link "This branch is 5 commits ahead of". Click on "5 commits ahead of" and make the pull request. Make sure you complete the exact instructions.

For the docs similar, but please make sure that in the Comparing changes window you correct manually the "Base repository" base to be "next" not "current" !!

Whet you're done, you can look at your PR and edit the first post to add the required PR links to connect the two to each-other.

alainstark, 2024-04-25:
Okay, I think I'm almost there.

I still have a problem with updating CODEOWNERS.
I am not the principal writer of this component. I have only adapt this to support **ds2482-800** the principal writer is [mkncj](https://github.com/mknjc).
[mkncj/esphome](https://github.com/mknjc/esphome)

I wait for the approwing review.

nagyrobi, 2024-04-25:
You can safely put yourself as codeowner because the goal is to have the system notify somebody if the component needs some changes in the future, that's all.

alainstark, 2024-04-25:
Ok I have do it.

alainstark, 2024-04-26:
I think I didn't do the right thing regarding `CODEOWNERS`.
In the [PR](https://github.com/esphome/esphome/pull/6616), the `needs-codeowners` label is always present.
I have changed __init__.py and run script/build_codeowners.py
The script did not give me any errors.
I then wanted to commit the modification (in __init__.py) which restarted the verification tool which went into error. So, I went back.
What should I do ?

nagyrobi, 2024-04-26:
Maybe the script is broken. Let your codeowners in, I removed the label manually.

alainstark, 2024-04-26:
Ok I added the line again that give me an error in script/ci-custom :

```
 . venv/bin/activate
  script/ci-custom.py
  script/build_codeowners.py --check
  shell: /usr/bin/bash -e {0}
  env:
    DEFAULT_PYTHON: 3.9
    PYUPGRADE_TARGET: --py39-plus
    pythonLocation: /opt/hostedtoolcache/Python/3.9.19/x64
    PKG_CONFIG_PATH: /opt/hostedtoolcache/Python/3.9.19/x64/lib/pkgconfig
    Python_ROOT_DIR: /opt/hostedtoolcache/Python/3.9.19/x64
    Python2_ROOT_DIR: /opt/hostedtoolcache/Python/3.9.19/x64
    Python3_ROOT_DIR: /opt/hostedtoolcache/Python/3.9.19/x64
    LD_LIBRARY_PATH: /opt/hostedtoolcache/Python/3.9.19/x64/lib
CODEOWNERS file is not up to date.
Please run `script/build_codeowners.py`
Error: Process completed with exit code 1.
```

nagyrobi, 2024-04-27:
It's fine

alainstark, 2024-04-30:
Hello,
Is there anything left to do or should I just wait ?

nagyrobi, 2024-05-01:
Just wait patiently from now...

szogun1987, 2024-10-08:
Feature request #621 is about the same device.
Comment just to track.

tomtom13, 2025-07-17:
> Feature request [#621](https://github.com/esphome/feature-requests/issues/621) is about the same device. Comment just to track.

Yeah, but that got closed by a MR that only implemented ONE variant 2484. All other variant work with just ONE bus.

mrangics, 2025-12-19:
Try this one_
https://github.com/mrangics/ds2482