esphome/feature-requests #2695: Generic hygrostat

mihaimdinca, 2024-04-20:
**Describe the problem you have/What new integration you would like**

A hygrostat component, similar to the generic hygrostat integration from home assistant, would allow the use of a single device (like, for example Sonoff THR 316) in conjunction with a dumb dehumidifier and have it integrated in home assistant but also having it running independent if home assistant stops running for some reason or the connection is lost.
**Please describe your use case for this integration and alternatives you've tried:**

I believe that the component would be very similar to the thermostat but oriented towards managing humidity rather than temperature.
**Additional context**

paravoid, 2024-05-04:
Duplicate of #1859?

closed 2024-05-04 by nagyrobi

Pastitas, 2024-05-22:
This is a bang bang controller for humidity, i dont think it's a duplicate. Has this been implemented then?