esphome/feature-requests #2696: Add support for Waveshare 3.52inch e-Paper raw display( 360 × 240, SPI Interface)

acfnews, 2024-04-21:
**Describe the problem you have/What new integration you would like**
For implementing an existing Waveshare e-paper into a modified Squeezebox Radio. The whole set should be then capable of being an announcer, display and even microphone for local speech recognition.

**Please describe your use case for this integration and alternatives you've tried:**
I woul
**Additional context**

wrcrooks, 2024-08-01:
I'm trying to get my development environment set up right now. I believe it's a type B screen so it should be relatively easy to create a profile for it. Also, I'm willing to test if anyone else beats me to getting a binary with support.

My current .yaml settings for my testing board:
```
spi:
  clk_pin: 13
  mosi_pin: 14

display:
  - platform: waveshare_epaper
    cs_pin: 15
    dc_pin: 27
    busy_pin: 25
    reset_pin: 26
    model: 1.54in # need to create a profile to be able to compile with '3.52in'
    rotation: 90°
    update_interval: 3s
    reset_duration: 20ms
```

jazzmonger, 2025-01-31:
nobody working on this?  I have the same display & would like to use it...