esphome/feature-requests #2697: Add support for WeActStudio.EpaperModule: WeAct Studio 1.54 Inch Epaper Module EPD

lboue, 2024-04-22:
**Describe the problem you have/What new integration you would like**

Add support for WeActStudio.EpaperModule: WeAct Studio 1.54 Inch Epaper Module EPD

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

[WeAct-EpaperModule-1.54 Board Shape.pdf
](https://github.com/WeActStudio/WeActStudio.EpaperModule/blob/master/Hardware/WeAct-EpaperModule-1.54%20Board%20Shape%20%E5%A4%96%E5%BD%A2.pdf)

latonita, 2024-05-07:
@lboue Hi, did you try waveshare epaper component?
I suppose it shall work with ssd1681

latonita, 2024-05-20:
@lboue This display finally came to me from aliexpress... and it properly works with existing component.

I attached it to my esp32dev module like this:

```yaml

spi:
  clk_pin: GPIO18 # scl
  mosi_pin: GPIO23 # sda

display:
  - platform: waveshare_epaper
    model: 1.54inv2
    cs_pin: GPIO17
    dc_pin: GPIO22
    reset_pin: GPIO19
    busy_pin: GPIO21
    update_interval: 15s
    full_update_every: 30
    lambda: |-
      it.print(10, 10, id(font1), "Hello World!");
```

I might update documentation so it will include WeAct name so it can be found by search function on the site.

lboue, 2024-05-21:
@latonita 
Thanks for the test. I'm going to try it again.

closed 2024-11-18 by lboue