esphome/feature-requests #2700: Add attribute name to callback parameters for `CustomAPIDevice::subscribe_homeassistant_state`

olicooper, 2024-04-26:
**Describe the problem you have/What new integration you would like**

I am finding that I am having to add a wrapper function for my state subscription callbacks because the `subscribe_homeassistant_state` callback doesn't return the attribute name as a parameter.

I would like the function changing (or a new one adding) from this:
e.g. `this->subscribe_homeassistant_state([this](std::string entity_id, std::string state) { }, "light.my_light", "supported_modes");`
```
void APIConnection::on_home_assistant_state_response(const HomeAssistantStateResponse &msg) {
  for (auto &it : this->parent_->get_state_subs()) {
    if (it.entity_id == msg.entity_id && it.attribute.value() == msg.attribute) {
      it.callback(msg.state);
    }
  }
}
```
To something like this
e.g. `this->subscribe_homeassistant_state([this](std::string entity_id, std::string attribute, std::string state, std::string attribute) { }, "light.my_light", "supported_modes");`
```
void APIConnection::on_home_assistant_state_response(const HomeAssistantStateResponse &msg) {
  for (auto &it : this->parent_->get_state_subs()) {
    if (it.entity_id == msg.entity_id && it.attribute.value() == msg.attribute) {
      it.callback(msg.state, msg.attribute);
    }
  }
}
```

**Please describe your use case for this integration and alternatives you've tried:**

Currently, I am creating lots of wrapper functions ([seen here](https://github.com/olicooper/esphome-nspanel-lovelace-native/blob/6a71629c1cb14a3f919b14381e67ea1f1e8bc9be/components/nspanel_lovelace/nspanel_lovelace.cpp#L1228-L1293)) to give me the functionality but I have a lot more to add and it is getting messy now.

**Additional context**

ssieb, 2024-04-26:
Why do you need to make a custom component like this?  What are you trying to do?

olicooper, 2024-04-27:
The project can be found here which will explain the idea better: [esphome-nspanel-lovelace-native](https://github.com/olicooper/esphome-nspanel-lovelace-native)

Essentially my project is an interface between a screen and Home Assistant. To build the commands for the screen I need to subscribe to state/attribute updates from HA entities that don't exist within the device, and because esphome 'components' represent internal device features which are exposed to HA, they don't work for what I need - which is a simple way to ask HA what the current state is of arbitrary entities.

There are many external entities and attributes that I need to monitor and creating a new function for every attribute type is not feasible. If the callback could include the attribute name I would then be able to remove all the additional functions [mentioned here](https://github.com/olicooper/esphome-nspanel-lovelace-native/blob/6a71629c1cb14a3f919b14381e67ea1f1e8bc9be/components/nspanel_lovelace/nspanel_lovelace.cpp#L1228-L1293)