esphome/feature-requests #2701: modbus: Custom command without CRC

skinkie, 2024-04-26:
**Describe the problem you have/What new integration you would like**
I am using a non-modbus RS485 device which has a custom protocol. I would like te reuse the existing modbus integration but with the payload exactly as provided, and parsing the data using a lambda function.

Since disable_crc already exists for receiving messages (ignore_crc would have been a better description) having the ability not to append it does not seam to far fetched. https://github.com/esphome/esphome/blob/dev/esphome/components/modbus/modbus.cpp#L212

**Please describe your use case for this integration and alternatives you've tried:**
Reimplementing this device using the UART interface seems to be too 'low-level', I would like to have the flow control and message handling from the existing modbus controller code.

An another alternative would be to actually implement this protocol for this range of devices. It works in a similar way as modbus.

**Additional context**
The device I am trying to interface with is a Depolox RESIDUAL ANALYZER.

nagyrobi, 2024-04-27:
Maybe you could look at the pzem energy meters. One of them is actually modbus, but with a dedicated implementation.

Nistp, 2024-05-02:
@skinkie if you could provide me with the RS485 manual, which I can't seem to find on Evoqua's website for some reason, I'd be happy to take a stab at implementing the protocol with UART as a back-end.

skinkie, 2024-05-02:
@Nistp at this point we have quite some difficulty to get our device talking at all. It does not even reply with a simple python version.

Nistp, 2024-05-03:
> @Nistp at this point we have quite some difficulty to get our device talking at all. It does not even reply with a simple python version.

I'm sorry to hear that :(

Nistp, 2024-05-03:
@skinkie I'm sorry to hear that :(