esphome/feature-requests #2704: Playing a embedded audio file

ChrisOboe, 2024-04-28:
**Describe the problem you have/What new integration you would like**

Currently media player is only able to play media from an url. I'd like to play media from an embedded file (e.g. in a byte array). 

**Please describe your use case for this integration and alternatives you've tried:**

I've built a door bell based on esphome with i2c_audio and media player and and amplifier. It works fine as long as the device has network connection. But i need a fallback, to play a sound even if there is no network connection. 

As a workarround i will try to use speaker play to play raw audio. But in that case i assume i will loose the codec support of media player. So the audio will propably be very limited by the space available.

**Additional context**

nagyrobi, 2024-04-29:
A config similar to `img` would be useful, referencing a file to be embedded in the binary.

docteurzoidberg, 2024-05-26:
definetly need embedding sound for my projects. the actual uses of dfplayer seems obsolete for low bitrate sounds / just a fancy bip or startup sound

nagyrobi, 2024-06-14:
Try @jesserockz 's custom component for this purpose: https://github.com/jesserockz/esphome-components/tree/main/components/file

Maybe it would be worth including it in the main code...

jhbruhn, 2024-07-18:
But this can not be used with the media_player component, only with a speaker component. There can be cases where no speaker but a media_player is configured, for example with [esphome_audio](https://github.com/gnumpi/esphome_audio).

closed 2025-02-10 by jesserockz