esphome/feature-requests #2705: Add high pass filter to sensor component

alex-masetti, 2024-04-28:
**Describe the problem you have/What new integration you would like**
There isn’t a high pass filter available for the esphome sensor component. Propose that one be added. Could be as simple as the example below.

Parameters:
- time_constant
- update_period
- `alpha = time_constant/(time_constant+update_period)`

Difference Equation
`output = alpha * ( previous_output + input - previous_input)`

**Please describe your use case for this integration and alternatives you've tried:**
Attempting to use an accelerometer to detect vibration and would like to be able to filter out DC accelerations (gravity, sensor bias) without having to calibrate or calculate the orientation of the sensor.

**Additional context**
N/A

latonita, 2024-05-07:
That's a good idea. Though, it only needs `alpha` as a parameter - it would be easy to use.

pricard32, 2025-05-25:
I do need this one as well, Low Pass and Hi Pass could be very useful.
I can do it in HA, but would prefer to get the filtered data coming out of the ESP itself.