esphome/feature-requests #2706: Add Support For Tuya NDIR Co2 Sensor MTP40-F

plplaaa2, 2024-04-29:
**Describe the problem you have/What new integration you would like**

Add Support For Tuya NDIR Co2 Sensor MTP40-F

[image]

[image]

**Data Sheet**

https://nettigo.eu/attachments/608

MTP40-F sensor is almost the same as mhz-19.<br>

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

**Arduino library for MTP40F CO2 sensor.**

https://github.com/RobTillaart/MTP40F

Nistp, 2024-05-03:
I'll be happy to work on this, but I'd need someone with the sensor to test it for me.

plplaaa2, 2024-05-03:
> I'll be happy to work on this, but I'd need someone with the sensor to test it for me.

I think I can help the test. But I'm not a programmer, so I think only limited tests are possible.

As a result of my own test with the mhz-19 component, I was able to read the co2 value. 
However, ABC or calibrate zero does not seem to work.

[image]

[14:49:10][VV][uart.arduino_esp32:180]:     Flushing...
[14:49:10][D][mhz19:065]: MHZ19 Received CO₂=538ppm Temperature=-40°C Status=0x00
[14:49:10][V][sensor:043]: 'CO2 Level': Received new state 538.000000
[14:49:10][D][sensor:094]: 'CO2 Level': Sending state 538.00000 ppm with 0 decimals of accuracy
[14:49:10][VV][api.service:140]: send_sensor_state_response: SensorStateResponse {
  key: 822384752
  state: 538
  missing_state: NO
}

Nistp, 2024-05-06:
That's pretty interesting, as the commands seem to be quite different for the two, based off the datasheet! I'll start working on this, it might take me a bit. I'll keep you updated!