esphome/feature-requests #2710: automatic retry for OTA-updates

gon0, 2024-05-03:
**Describe the problem you have/What new integration you would like**
I would wish for a ESPHome OTA retry-function. A counter could be used, which I would set e.g. to “50”, which automatically clicks “retry” for me, if an OTA update fails. By standard, it is set to “1”.
[image]

**Please describe your use case for this integration and alternatives you've tried:**
Problem in my scenario: I have some quite packed ESP32 running wih a lot of peripherals. This causes the OTA-update to fail most of the time. After retrying ~15 times the OTA update succeeds. I added a collection of the various error messages I receive in the code section below.

I already implemented [Safe Mode Button — ESPHome](https://esphome.io/components/button/safe_mode.html) , but It did not help for having a successful OTA update on the first try (it also takes > 10 trys, until an OTA update succeeds).

The Wifi-connection seems to be good (Fritzbox 6490 Cable 2 m away from ESP32).

**Additional context**
An excerpt from console outputs from failed OTA updates:

```
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.3
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Wire @ 2.0.0
|-- ArduinoJson @ 6.18.5
|-- Improv @ 1.2.3
|-- FastLED @ 3.3.2
RAM:   [==        ]  16.5% (used 53944 bytes from 327680 bytes)
Flash: [========= ]  92.5% (used 1697605 bytes from 1835008 bytes)
========================= [SUCCESS] Took 10.38 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.122.162
INFO Uploading /data/build/esphome-web-45e2b0/.pioenvs/esphome-web-45e2b0/firmware.bin (1703376 bytes)
Uploading: [                                                            ] 0% ERROR Error receiving acknowledge chunk OK: [Errno 104] Connection reset by peer

or

WARNING esphome-web-45e2b0 @ 192.168.122.162: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for esphome-web-45e2b0 @ 192.168.122.162
WARNING Disconnected from API

or 

ERROR Error receiving acknowledge chunk OK: [Errno 104] Connection reset by peer

or 

Uploading: [=                                                           ] 1% ERROR Error receiving acknowledge chunk OK: [Errno 104] Connection reset by peer

or 

ERROR Error receiving acknowledge version: timed out

or

Uploading: [=========                                                   ] 14% ERROR Error receiving acknowledge chunk OK: timed out

or

ERROR Error binary size: Unknown error from ESP
```
Just for information, this is the ESPHome-yaml file for the ESP32:

```
esp32:
  board: wemos_d1_mini32
  framework:
    type: arduino
[... 268 lines cut ...]
  - platform: partition
    name: "Indicator top right"
    default_transition_length: 0s
    icon: "mdi:arrow-top-right-bold-outline"
    segments:
      - id: led_Taster_5mm_LED_Lights
        from: 5
        to: 5                             
  - platform: fastled_clockless
    chipset: WS2812B
    pin: GPIO33
    num_leds: 6
    rgb_order: RGB
    name: "led_Taster_5mm_LED"
    id: led_Taster_5mm_LED_Lights
    default_transition_length: 0s
    color_correct: [50%, 50%, 50%]
    restore_mode: ALWAYS_ON
  - platform: binary
    name: "LED button left"
    output: LED_button_left_output
  - platform: binary
    name: "LED button right"
    output: LED_button_right_output
```

pvoo, 2024-09-08:
This would be a good addition, i'm seeing similar results where OTA often fails due to _ERROR: receiving acknowledge chunk OK: timed out_

dothebart, 2025-06-21:
whenever you have devices with extended sleep intervals this would be a good thing to have.