esphome/feature-requests #2713: Kubernetes deployments - helm charts

jlpedrosa, 2024-05-04:
**Describe the problem you have/What new integration you would like**
I would like to deploy esphome in kubernetes. I have been contributing to different projects creating helm charts to deploy them in k8s. Zigbee2mqtt, some for home assistant.. etc.

**I Can create the PR for it, if it would be accepted.**

**Please describe your use case for this integration and alternatives you've tried:**
Running it as containter, but is a pain to update containers manually.

**Additional context**
I'm running the hole homeassistant ecosystem inside k8s, this is the missing piece.

nagyrobi, 2024-05-05:
Need exact documentation too

jlpedrosa, 2024-05-06:
Hi @nagyrobi  @jesserockz, I understood that this was a desired feature-request, so I created the PR for it and got inmediatelly closed. 

Let me know if you really want it so I can reopen the PR or close this feature-request.

closed 2024-05-06 by nagyrobi