esphome/feature-requests #2720: Addition of time based valve, similar to time based cover

balk77, 2024-05-19:
I am using a valve with 3 wires: 2 wires to control direction and a common neutral. Powering one wire turns the motor one way, power on the other turns the other way. I am using a Sonoff Dual to control the wires. The current setup is using[ time based cover](https://esphome.io/components/cover/time_based); it takes 16 seconds to close or open in my case; the component estimates the position based on run time. Of course there is some drift but that does not really matter; valve is closed every now and then to reset the position.

It would be nice to have a time based [valve component](https://esphome.io/components/valve/) as well; I guess that there are more valves out there that work on this principle (more than there are affordable valves with true position feedback).

nagyrobi, 2024-05-21:
@kbx81 :)

kraus-wagner, 2024-05-26:
I would really appreciate that too. I control a heating mixer valve. The running time is 2 minutes in each direction.

Aidam, 2024-12-08:
Was looking for the  same thing - Glad i found this post, at least i know i am not missing anything