esphome/feature-requests #2722: implement a lightweight Matter over WiFi protocol stack in ESPHome

lboue, 2024-05-21:
**Describe the problem you have/What new integration you would like**

Would it be possible to implement a lightweight Matter over WiFi protocol stack in ESPHome?

**Please describe your use case for this integration and alternatives you've tried:**

The aim is to be able to control these devices directly via a Matter controller.

**Additional context**

Hedda, 2024-05-21:
Duplicate of https://github.com/esphome/feature-requests/issues/1430

lboue, 2024-05-21:
> Duplicate of #1430 as that applies to Matter over IP on both Wi-Fi and Ethernet

This issue #1430 was opened in 2021 and is becoming a catch-all.

Hedda, 2024-05-21:
> This issue #1430 was opened in 2021 and is becoming a catch-all. There is a discussion about Zigbee, Thread Border Router, etc.

I disagree, I think https://github.com/esphome/feature-requests/issues/1430 is clearly only about connectivity using the high-level Matter standrard (and not really about Thread).

You might be thinking about https://github.com/esphome/feature-requests/issues/1397 which is a discussion about OpenThread and Zigbee on ESP32 (like ESP32-H2/ESP32-C6)

lboue, 2024-05-21:
> > This issue #1430 was opened in 2021 and is becoming a catch-all. There is a discussion about Zigbee, Thread Border Router, etc.
> 
> I disagree.
> 
> #1430 is only about connectivity using the Matter application-layer, (it is not really about media-layer, e.g. Thread/OpenThread).

Border Router
https://github.com/esphome/feature-requests/issues/1430#issuecomment-1010939789
https://github.com/esphome/feature-requests/issues/1430#issuecomment-1268384267
https://github.com/esphome/feature-requests/issues/1430#issuecomment-1259198292

OpenThread
https://github.com/esphome/feature-requests/issues/1430#issuecomment-1268384267
https://github.com/esphome/feature-requests/issues/1430#issuecomment-1259198292
https://github.com/esphome/feature-requests/issues/1430#issuecomment-1435639413

Zigbee
https://github.com/esphome/feature-requests/issues/1430#issuecomment-1010939789
https://github.com/esphome/feature-requests/issues/1430#issuecomment-1259284389
https://github.com/esphome/feature-requests/issues/1430#issuecomment-1268384267
https://github.com/esphome/feature-requests/issues/1430#issuecomment-1435639413

> Are you thinking about #1397? That is a discussion about OpenThread media-layer (and Zigbee) on ESP32 (like ESP32-H2)
No

nicx, 2025-05-05:
waiting for matter support too... is this somehow in development?

mhalano, 2025-05-05:
I think we need to finish Thread support first. I know that can be a WI-Fi-only implementation, but that's not the standard.

nicx, 2025-05-05:
@mhalano why should Wi-Fi not be the standard? Both protocols are completly standardized with Matter. In addition to these 2 wireless protocols also Ethernet is included in the matter standard.

Hedda, 2025-05-05:
Again this is another reason why this is a duplicate of this other request:

- https://github.com/esphome/feature-requests/issues/1430

Anyway, Matter of WiFi follows the standard, Matter for Ethernet follows the standard, and Matter over Thread follows the standard, there is nothing in the standards that you need to support more than one of those transport protocols.

mhalano, 2025-05-06:
What I mean, is: ESPHome is not a finished product, like my smartplug that just supports Wi-Fi, but it's a platform. We need to implement Matter with full support because someone want to use Wi-Fi, but other folk will want to use Thread, and a third one will want Ethernet. Said that, I also think we could have a preview implementation just with Wi-Fi, that, I think, is the most used system. But to consider the solution stable and ready, we need all three implemented. Again, ESPHome isn't a finished product, but a platform.

Hedda, 2025-05-06:
Hence why this is a duplicate of https://github.com/esphome/feature-requests/issues/1430

goloveychuk, 2025-07-12:
Fyi thread is implemented in esphome