esphome/feature-requests #2725: Wiegand RFID Reader handling 7byte UID

pvtex, 2024-05-21:
**Describe the problem you have/What new integration you would like**
when using 7byte RFID card which only have UID changes in the first bytes all card have the same UID
because only 4 bytes fromthe UID are used

**Please describe your use case for this integration and alternatives you've tried:**
option to select how a 7 byte UID is handled (first or last 4 bytes)

**Additional context**

nagyrobi, 2024-05-21:
@ssieb

ssieb, 2024-05-21:
The component can read up to 64 bits.  What are you seeing in the log?
I suspect it's more likely that your reader is only sending the first bits.

pvtex, 2024-05-22:
Ok.
I think it is my reader only supporting wiegand34.
I have order a new one with up to 58bit.

Request can be closed.

closed 2024-05-22 by pvtex

ssieb, 2024-05-22:
When you get it, create an issue with details so that I can add support for that data format.