esphome/feature-requests #2726: Modbus TCP Server support

RoganDawes, 2024-05-23:
As originally requested in #708, it would be very useful to be able to interact with a modbus server using modbus TCP, ideally at the same time as ESPHome is polling it.

In other words, if there is a UART connected to a modbus server, or esphome is acting as a modbus server itself, it would be useful to be able to interface with it using modbus TCP.

Obviously modbus RTU queries (if there are any) would have priority due to latency requirements.

External modbus TCP queries would need to be interleaved with internal ones such that neither gets corrupted.

I have an esp32 connected to my inverter, and have defined a number of entities the esphome queries. However, it would be useful to allow ad hoc queries of registers that have not already been defined, without having to reflash new firmware.

This can allow eg manufacturers tooling to interact with the device at the same time as esphome does.

I have used stream_server to expose the uart, but that is an exclusive use only.

**Additional context**

nagyrobi, 2024-05-23:
A proper master and slave (in newer naming client and server) implementation for ModBUS would be nice.

My inverters can be queried via TCP too. That would relief the uarts free for more, optional factory peripherals to be connected.

Justheretoreportanissue, 2024-07-09:
+1
I have my inverter installed remote.
Inverter have Modbus TCP.
Remote location have network, ethernet and WiFi.
Remote location can access the Internet.

I would love to have an esphome device connect to my inverter over Modbus TCP and then act as a gateway to some 'public' MQTT broker over TLS on the Internet.
Home Assistant would then be able to get all values from my inverter by reading them from the MQTT broker on the Internet.

I have already verified that the esphome device can access the MQTT broker on the Internet (EMQ) and write to topics when at the remote location.

Adding support for Modbus TCP would make my day/week/month/year/millennium and complete the missing link in my config.

Having a remote location where you would like to communicate to something with Modbus TCP and have it report back to your HA can not be that uncommon, right? 

Regards,

   /Jonas...

RoganDawes, 2024-07-09:
> +1 I have my inverter installed remote. Inverter have Modbus TCP. Remote location have network, ethernet and WiFi. Remote location can access the Internet.
>
> I would love to have an esphome device [...]

Please note that this is NOT the same as the other issue that you also just commented on. This issue is asking for a modbus TCP *server* implemented in ESPHome, not a Modbus TCP *client* which the other issue refers to.

Waz-Cpt, 2024-08-19:
+1 for a Modbus TCP gateway feature.

It would be awesome to see is the following could aid in the development.
[https://github.com/zivillian/esp32-modbus-gateway](url)

madejackson, 2024-10-23:
+1, Would be very useful. Home assistant modbus integration is laughably slow (1s polling rate at most) and only usable for monitoring, Realtime applications like lighting and blinds control are impossible to realize with 1s delay (would need to be somewhere in the 100ms range).

mortenx, 2024-11-22:
i made modbustcp with lan/esphome/mqtt
https://www.instructables.com/MODBUS-TCP-ESPHOMEmqtt-Gateway/

It would be nice if there was a wifi option too, I don't know how to do that.

mortenx, 2024-11-26:
is it possible to use something like this?

modbus:
  - platform: tcp
    id: modbus_tcp1
    host: 192.168.1.50  # Replace with your Modbus TCP device IP
    port: 502           # Default Modbus TCP port

modbus_controller:
  - type: tcp
    id: modbus_tcp1
    host: 192.168.1.50  # Replace with your Modbus TCP device IP
    port: 502           # Default Modbus TCP port

modbus_controller:
  - id: modbus_tcp1
    address: 192.168.1.50  # IP address of your Modbus TCP device
    port: 502              # Default Modbus TCP port

RoganDawes, 2024-11-26:
> +1, Would be very useful. Home assistant modbus integration is laughably slow (1s polling rate at most) and only usable for monitoring, Realtime applications like lighting and blinds control are impossible to realize with 1s delay (would need to be somewhere in the 100ms range).

You can adjust the poll rate for certain registers, as far as I know. See https://esphome.io/components/modbus_controller.html#:~:text=update_interval

RoganDawes, 2024-11-26:
Thinking about the implementation of this a little more, I guess conceptually the TCP listener would just be another modbus "sensor", in that it would generate a modbus request, and receive a modbus response. While one modbus request is in flight, no others are submitted, which is exactly the behaviour that we want. The TCP listener would obviously be responsible for extracting the PDU from the modbus TCP APDU, and formulating the response, again, formatted appropriately.

Obviously, this would not be exposed to the API as an actual sensor, other than perhaps recording the number of messages passed through, but that is independent of the actual functionality.

It feels appropriate that the TCP "sensor" would ignore `update_interval`, though, in my opinion,otherwise things like e.g. a firmware update would take forever.

So, given the example from the modbus_controller page, I think configuration could look like:
```
modbus:
  - uart_id: uart_modbus_client
    id: modbus_client
    tcp_server:
      - port: 502 #default
      - max_connections: 1 #default
```
which would result in a TCP listener on the specified port.

It doesn't seem unreasonable to also do something like:
```
modbus:
  - modbus_server:
      - address: 192.168.1.101
      - port: 502 #default
    id: modbus_client
    tcp_server:
      - port: 502 #default
```
which would convert the modbus PDU's from any defined sensors into modbus TCP APDU's, send them to the defined modbus TCP server (at <address:port>), receive the answer, and update the sensor appropriately. Again, it might be useful to provide a multiplexing capability to allow an additional modbus client to relay their requests through our node, in the case where the target only allows one simultaneous connection.

Obviously, the component would be responsible for re-establishing a broken connection should it close/be reset for any reason, and perhaps backoff settings would be appropriate to prevent accidental DoS.

mortenx, 2024-11-26:
sensor like this?

sensor:
  - platform: modbus_controller
    modbus_id: modbus_tcp1
    name: "Temperature Sensor"
    register_type: holding  # Typically 'holding' or 'input', based on device specs
    address: 1              # Modbus register address for temperature
    value_type: U_WORD      # Data type for the register
    unit_of_measurement: "°C"
    update_interval: 10s

Waz-Cpt, 2024-11-26:
I believe thee is something a mis here.

There are two aspects to this concept of a Modbus TCP server.

One is to use the defined modbus RTU registers or coils and have them accessible as Modbus TCP from the IP of the ESPhome device on a defined port.

The other is a transparent Modbus RTU to TCP Gateway. It makes a lot of sense to have a transparent gateway as a component in ESPhome it has a set functionality.

Addons like KellerZAs Sunsynk/Deye addon could use this as their data source. This also means that you can have an easy to deploy Modbus RTU to TCP gateway for any RS485 Modbus RTU device.

It does not really make sense to define Modbus registers or coils on the ESP only to then have something communicate to it via Modbus TCP. Rather use MQTT or the native API to transmit the defines registers or coils.

What can make some sense is to have an ESP32 connecting to a Modbus TCP device to do independant control or automation without the need of Home Assistant. This is typically not referred to as a Modbus TCP Server.

If a device is already outputting Modbus TCP then you can handle this directly in Home Assistant or build a simple addon to transform the Modbus TCP device into devices in HA or Auto Discovered MQTT devices.

madejackson, 2024-11-26:
> What can make some sense is to have an ESP32 connecting to a Modbus TCP device to do independant control or automation without the need of Home Assistant. This is typically not referred to as a Modbus TCP Server.

Yeah it is. As you said yourself there is always a Server and a Client. We're talking here about ESPhome as a client talking to an external server.

> 
> If a device is already outputting Modbus TCP then you can handle this directly in Home Assistant or build a simple addon to transform the Modbus TCP device into devices in HA or Auto Discovered MQTT devices.

If you read my comment, you can see that the modbus-Integration of HA is lacking. Minimum Cycle is 1s which is just not really useful in many cases. Modern PLCs have cycle times as low as 50ms. Nobody want's to wait >1s for a light to turn on.

RoganDawes, 2024-11-26:
> I believe there is something a mis here. Typically, there is ether a Modbus TCP server or client.
> 
> There are two aspects to this concept of a Modbus TCP server.
> 
> One is to use the defined modbus RTU registers or coils and have them accessible as Modbus TCP from the IP of the ESPhome device on a defined port.
> 
> The other is a transparent Modbus RTU to TCP Gateway. It makes a lot of sense to have a transparent gateway as a component in ESPhome it has a set functionality.
> 
> Addons like KellerZAs Sunsynk/Deye addon could use this as their data source. This also means that you can have an easy to deploy Modbus RTU to TCP gateway for any RS485 Modbus RTU device.
> 
> It does not really make sense to define Modbus registers or coils on the ESP only to then have something communicate to it via Modbus TCP. Rather use MQTT or the native API to transmit the defines registers or coils.
> 
> What can make some sense is to have an ESP32 connecting to a Modbus TCP device to do independant control or automation without the need of Home Assistant. This is typically not referred to as a Modbus TCP Server.
> 
> If a device is already outputting Modbus TCP then you can handle this directly in Home Assistant or build a simple addon to transform the Modbus TCP device into devices in HA or Auto Discovered MQTT devices.

I think I explained the use case adequately in my original post. ESPHome does direct communication with the Modbus UART device, and presents entities to Home Assistant, but there may also be a need for e.g. vendor tools to access the device over TCP, for example, to update firmware, or even just to perform ad hoc queries whiile refining the ESPHome config. In this case, ESPhome would be acting as a Modbus Server.

And yes, there is another feature request that addresses modbus TCP Client functionality, but it seems to me that there is actually a fair amount in common between them, in particular *where* the configuration happens (since it is just a `role` that separates them), as well as the requirement for a modbus TCP request encoder/decoder and response encoder/decoder. 

Obviously, if only modbus TCP server is configured, only the request decoder and response encoder are actually needed, and similarly, if only modbus TCP client is configured, only the request encoder and response decoder would be required. But it seems pointless to do only half the job when you are right there, if only because testing should be possible in a standalone/loopback configuration, much as the original modbus_server code was tested in loopback of one UART to another on the same board.

Waz-Cpt, 2024-11-26:
Ok, the title and initial chatter was a little confusing. I had previously had talks with Rogan about this.

I have been hunting for someone to help me turn an existing ESP32 adruino project into a component for ESPhome to have a transparent Modbus RTU to TCP Gateway device. Would be very useful for the EDGE IoT style systems.

The polling rate does make what has been discussed quite practical.

mortenx, 2024-11-26:
If I have a device on a remote network that transmits data, such as temperature, via Modbus TCP, what options are there to receive it via WiFi using esphome/esp32 and send it to another network using MQTT?

Waz-Cpt, 2024-11-26:
Currently KellerZA's Sunsynk/DEYE addon does exactly that but within HA or standalone Docker image. IAMMETER also make use of Modbus TCP through HA addons. But, as pointed out, a polling rate of faster than 1s is basically essential.

RoganDawes, 2024-11-26:
> > What can make some sense is to have an ESP32 connecting to a Modbus TCP device to do independant control or automation without the need of Home Assistant. This is typically not referred to as a Modbus TCP Server.
> 
> Yeah it is. As you said yourself there is always a Server and a Client. We're talking here about ESPhome as a client talking to an external server.
> 
> > If a device is already outputting Modbus TCP then you can handle this directly in Home Assistant or build a simple addon to transform the Modbus TCP device into devices in HA or Auto Discovered MQTT devices.
> 
> The modbus-Integration of HA is lacking. Minimum Cycle is 1s which is just not really useful in many cases. ESPhome shouldn't have this limit (in theory). Modern PLCs have cycle times as low as 50ms. Nobody wants to wait >1s for a light to turn on.

Not sure how you conclude that the minimum cycle is 1s? The modbus_controller derives from PollingComponent, which is where the `update_interval` parameter goes. I just compiled a test firmware with an `update_interval` of 50ms, and it was happy to do so. Not sure how practical it will be, as this might starve the rest of the system, possibly including the WiFi or the watchdog. Granted, I haven't tried actually running it myself.

Then again, I'm not sure whether actions such as writing to a register to turn on a light would be constrained to the update_interval at all, or whether they would happen as soon as triggered. Unless you are thinking more of the light switch as being read from a modbus server, then triggering the "light turn on" action subsequently?

Waz-Cpt, 2024-11-26:
@RoganDawes the 1s is referring to the Native HA polling rate of Modbus TCP sensors.

Waz-Cpt, 2024-11-26:
@mortenx I now understand the desire of the orginal request from @RoganDawes and your ideas. It would also be really convenient for creating a small remote display using one of the ESP32 display devices and connecting to a Modbus TCP of a device like a compressor or even a VSD. You would get realtime data while also being able to have data recorded in HA database/historian.

mortenx, 2024-11-26:
The main problem would be that the modbus tcp device is somewhere on a different network than HA and esphome/esp32 would be a good gateway

Waz-Cpt, 2024-11-26:
The challenge that I have seen, is not all Modbus TCP devices support multiple connections.

I have seen a few of the Modbus RTU to TCP gateways supporting a multi-client server setup. How reliable they are is questionable. Actually purchased a PUSR USR-M300 to play wit for our compressor and chiller at the office. Hopefully going to start messing around with it next week.

RoganDawes, 2024-11-26:
> The challenge that I have seen, is not all Modbus TCP devices support multiple connections.
>
> I have seen a few of the Modbus RTU to TCP gateways supporting a multi-client server setup. How reliable [...]

This was explicitly one of the use cases I mentioned for supporting a Modbus TCP Server in ESPHome, along with a Modbus TCP client - that it would allow multiplexing requests to a device that doesn't allow more than one connection.

RoganDawes, 2024-11-26:
> @RoganDawes the 1s is referring to the Native HA polling rate of Modbus TCP sensors.

I still don't see this in the code. By default, the PollingComponent is initialised with a 60s `update_interval`, as per modbus_component/__init__.py, but it allows being set to any "positive integer". There is a `command_throttle` parameter, which is default 0ms. I don't see any other impediment to faster polling.

RoganDawes, 2024-11-26:
> If I have a device on a remote network that transmits data, such as temperature, via Modbus TCP, what options are there to receive it via WiFi using esphome/esp32 and send it to another network using MQTT?

@mortenx Right now, probably buy an existing commercial dongle that offers these features. However, if https://github.com/esphome/feature-requests/issues/1992 is implemented (possibly aligning with some of the suggestions I have made in this issue), then it should be possible.

Updates to Modbus entities would be mirrored to MQTT, and MQTT would enable triggering of commands/writes to modbus registers.

mortenx, 2024-11-27:
I made a modbustcp_esphome with custom component
https://github.com/mortenx/modbustcp_wifi_esphome/blob/main/modbusTCP.yaml

martgras, 2025-05-11:
A 'tcpuart' component is another option.  
It would enable modbus to utilze it. 
Additionaly it's not limited to modbus. 
serial over ip would be possible as well

Waz-Cpt, 2025-05-21:
> Very new and lightly tested modbus TCP custom component:
> 
> https://github.com/m66b/esphome-stream-server-v2

I am guessing this does not work as transparent Modbus TCP Gateway?

Will it only provide configured Modbus registers to clients?

M66B, 2025-05-21:
> > Very new and lightly tested modbus TCP custom component:
> > https://github.com/m66b/esphome-stream-server-v2
> 
> I am guessing this does not work as transparent Modbus TCP Gateway?
> 
> Will it only provide configured Modbus registers to clients?

Indeed, it is meant to publish sensor values, and not meant as a gateway.

I'm using it to read out an SDM120 energy meter with modbus RTU and publish it as an EM24 energy meter via modbus TCP. For example:

```
sensor:
  - platform: modbus_controller
    id: sdm120_voltage
    modbus_controller_id: controller_client
    name: "SDM120 Voltage"
    register_type: read #Function code 4
    address: 0x0000
    device_class: voltage
    state_class: measurement
    value_type: FP32
    accuracy_decimals: 1
    unit_of_measurement: V
    on_value:
      then:
        - lambda: id(tcp).setRegisterSint32(1, 3, 0x0000, x * 10, 2500);
```

Bascht74, 2026-09-29:
Two different requests are in this thread.

If the ESP is itself the Modbus server, that server can be reached over Modbus TCP without a second controller. `tcp_uart` with `role: server` and `protocol: modbus` is only the socket. The server stays `modbus_server`.

https://github.com/Bascht74/esphome-tcp-uart

The other case is not covered. A TCP client and the local `modbus_controller` cannot share one UART. `uart_tcp` would be the only master on the pins, so the local sensors stop. Release v1.0.0, ESPHome 2026.8 or newer.