esphome/feature-requests #2730: Hx711 and deep sleep

Tomaszek77, 2024-05-25:
**Describe the problem you have/What new integration you would like**

It would be great to add feature to HX711, to allow putting this module to sleep before whole device goes to deep-sleep. It is possible in Arduino by putting CLK pin to high for 50ms.

**Please describe your use case for this integration and alternatives you've tried:**

There is no way to save on battery power without putting hx711 to sleep, without manipulating CLK Pin.

**Additional context**

In Arduino there is 
LoadCell.powerDown()

DjordjeMandic, 2025-01-25:
Since there is no way to add callback to powerdown function, when this https://github.com/esphome/esphome/pull/8138 gets merged, im gonna create action to put module into power down

DjordjeMandic, 2025-05-25:
If anyone is able to test this PR: https://github.com/esphome/esphome/pull/8899, feedback would be greatly appreciated.