esphome/feature-requests #2732: Add option to not prepend friendly_name to Home Assistant entities

DrBlokmeister, 2024-05-26:
**Describe the problem you have/What new integration you would like**

When adding a friendly_name to a device, this name is displayed in the ESPHome dashboard, improving user friendliness. However, this also prepends the friendly_name to entities in Home Assistant. When using a configuration like:
```yaml
substitutions:
  device_name: "Plug Electric Car"

esphome:
  friendly_name: ${device_name}

sensor:
  - platform: total_daily_energy
    name: "${device_name} Energy Today"
    power_id: "power"
```
Home Assistant will add the entity sensor.plug_electric_car_plug_electric_car_energy_today, which is unwanted.

It would be nice if we got the option to disable prepending of the friendly_name to any entities in Home Assistant.

**Please describe your use case for this integration and alternatives you've tried:**

Currently two options are available: 
- not using friendly name, which makes the ESPHome dashboard look more ugly.
- Not manually prepending the device name to the entities. However, when I have many entities referring to the same packaged file (such as network diagnostics) and these entities do not have a friendly name defined, suddenly the naming of the sensor would just be sensor.wifi_signal for a random device.

**Additional context**

I was a bit unsure if this is an ESPHome feature request or a Home Assistant feature request. If I'm in the wrong category, please let me know.

paravoid, 2024-06-15:
+1 this is kind of annoying :) Also see #2476 and in particular https://github.com/esphome/feature-requests/issues/2476#issuecomment-1844034117  which I believe is describing the same issue (although I find your example quite helpful in describing this too).

caco3, 2025-07-06:
As I have just pointed out in https://github.com/esphome/feature-requests/issues/2476#issuecomment-3042901117, the duplication seems to depend on the used view...

mateuszdrab, 2026-11-16:
This is very annoying, and as I'm migrating into ESPhome from Tasmota, I wanted to address this by designing a new naming strategy. 
I already decided that `name` will be `esphome-abcdef` based on the MAC address - need to keep those globally unique due to shared/relayed MDNS namespace between two homes.
As for friendly name, the same issue applies to Z2M entities, they all have area name in them, so it results in the duplicate. Though, in some instances, I noticed HA would magically deduplicate the area name out of the friendly name and its respective entity ID, but I still haven't figured out why it happens.
I'd personally prefer for HA to strip area name out of friendly names automatically, so that all devices can benefit from this.
If implemented in ESPhome, it leaves the same issue in Z2M and wherever there might be identical device names if they're deployed per area.
Having an area qualified friendly name in ESPhome allows for easier identification in the build history and its own web UI, but in HA, I simply might have to rename the device to remove the area from it. It seems like the only solution for now.