esphome/feature-requests #2733: FabGL (+lvgl ?)

docteurzoidberg, 2024-05-26:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

nagyrobi, 2024-05-26:
> FabGL library support (which have lvgl support) would be great.

Not sure how that works, but likely it's about LVGL's own video driver talking directly to FabGL. That's not how ESPHome works. ESPHome has its own display drivers - completely independent from LVGL (you can use them with lambdas too), and LVGL talks to these.

It would be needed to implement a FabGL mod integrating under ESPHome as a display driver (pretty much like LibreTiny is), which would be usable either with lambdas, LVGL or anything else the future may bring.

docteurzoidberg, 2024-05-27:
> > FabGL library support (which have lvgl support) would be great.
> 
> Not sure how that works, but likely it's about LVGL's own video driver talking directly to FabGL. That's not how ESPHome works. ESPHome has its own display drivers - completely independent from LVGL (you can use them with lambdas too), and LVGL talks to these.
> 
> It would be needed to implement a FabGL mod integrating under ESPHome as a display driver (pretty much like LibreTiny is generally), which would be usable either with lambdas, LVGL or anything else the future may bring.

Yes that would need a rewrite. but what i mean't was the porting of the VGAbuffers class and CVBS ones. could be called 'fabgl" compatible if drawing api is also implement but wouldn't be needed