esphome/feature-requests #2742: sftp "flash on boot"

otto2704, 2024-05-30:
**Describe the problem you have/What new integration you would like**

It would be cool, if esphome devices could receive images using sftp like boot images for PCs or raspberries.
For example if a change in the code makes the node rebooting continuously, this would be a possibility to re-flash them even if the yaml itself is having problems.
Of course, a dhcp-server and tftp-server would be necessary and properly configured.
**Please describe your use case for this integration and alternatives you've tried:**

I have many nodes which are built into appliances and within walls and it is hard to get usb-access to them.
For example I just uploaded a binary of a yaml which makes the node reboot as soon as it starts and wireless flashing is not working.
So instead of dismounting the node it would be cool just to place the binary on a tftp server and it would be written to the node as soon as it boots.
Maybe some kind of memory ("already flashed") would be cool, so the binary would only be flashed one time and not if it was already flashed and is active on the node.
**Additional context**

randybb, 2024-05-30:
You can achieve the same with https://github.com/esphome/esphome/pull/5586

otto2704, 2024-05-30:
@randybb 
Thanks for your comment!
I am not really sure and maybe I did not really understand the http_request, but how could this help when a binary is not really booting anymore? As far as I read it, the binary can be uploaded using a button or mqtt or whatever, but after booting?

nagyrobi, 2024-05-30:
You have safe mode for that. By default, if it enters in a bootloop, [after 10 quick boot cycles](https://esphome.io/components/ota.html#ota-update-component) it starts a safe mode, which allows you to flash via ota again.

otto2704, 2024-05-30:
Ah ok. I did not know this.
So I think, I will mark this as closed.
Thank you!

randybb, 2024-05-30:
Yes, there is safe mode, so you can still flash your device via OTA (`platform: esphome`). The new component can give you that auto-update option, so your device can check if there is a new binary in a repo.

closed 2024-05-30 by otto2704