esphome/feature-requests #2744: Support ST7701 type 5 init operations

Kernald, 2024-06-03:
**Describe the problem you have/What new integration you would like**
ESPHome currently supports custom init operations and ST7701 type 1 (with type 7 seemingly implemented but commented out and untested: https://github.com/esphome/esphome/blob/b8d2a6f574e4f49d12d3e677220c9dd92a364aae/esphome/components/st7701s/init_sequences.py#L360)

**Please describe your use case for this integration and alternatives you've tried:**
This is required for some specific displays, e.g. [this one](https://wiki.makerfabs.com/MaTouch_ESP32_S3_2.1_Rotary_TFT_with_Touch.html).

**Additional context**
Type 5 is defined here: https://github.com/moononournation/Arduino_GFX/blob/b6aa7551a6466182f12810d030e4c74e28673cad/src/display/Arduino_RGB_Display.h#L665

Krastanov, 2024-11-22:
This is about ST7701S, not about ST7701, right? (these are different chips)

clydebarrow, 2024-11-24:
You can specify that in your yaml using the `init_sequence:` config.