esphome/feature-requests #2745: Support for Display Driver: EK9716BD3 & EK73002ACGB

lboue, 2024-06-04:
**Describe the problem you have/What new integration you would like**

Support for Display Driver: EK9716BD3 & EK73002ACGB

**Please describe your use case for this integration and alternatives you've tried:**

[CrowPanel 7.0" -HMI ESP32 Display 800x480 RGB TFT LCD Touch Screen](https://www.elecrow.com/esp32-display-7-inch-hmi-display-rgb-tft-lcd-touch-screen-support-lvgl.html)

**Additional context**

lboue, 2024-06-04:
I'm currently testing with this with ESPHome [RPI_DPI_RGB Display Driver](https://esphome.io/components/display/rpi_dpi_rgb#rpi-dpi-rgb-display-driver) component. I hope it will work but I'm not sure.

lboue, 2024-07-08:
I've finally managed to get something on my [Elecrow CrowPanel 7.0"](https://www.elecrow.com/esp32-display-7-inch-hmi-display-rgb-tft-lcd-touch-screen-support-lvgl.html):

```
#-------------------------------------------
# Touchscreen gt911 i2c
#-------------------------------------------
touchscreen:
  platform: gt911
  id: my_touchscreen
  display: my_display
  interrupt_pin: 38

i2c:
  sda: 19
  scl: 20
  scan: true

display:
  - platform: rpi_dpi_rgb
    id: my_display
    data_pins:
      red:
        - 14
        - 21
        - 47
        - 48
        - 45
      green:
        - 9
        - 46
        - 3
        - 8
        - 16 
        - 1
      blue:
        - 15
        - 7
        - 6
        - 5
        - 4
    de_pin: 41
    hsync_pin: 39
    vsync_pin: 40
    pclk_pin: 0

    hsync_front_porch: 40
    hsync_pulse_width: 48
    hsync_back_porch: 13

    vsync_front_porch: 1
    vsync_pulse_width: 31
    vsync_back_porch: 13

    pclk_inverted: true
    pclk_frequency: 8500000

    color_order: RGB

    auto_clear_enabled: false
    update_interval: 5s
    dimensions:
      width: 800
      height: 480
```

koma6379, 2024-08-31:
Hello,
I have a question,
can you post the whole yaml?
Thanks Matthias

boussouira, 2024-09-01:
@lboue Do other things work on this CrowPanel 7.0 board? It's by far the cheapest and seem well made

AndyDann, 2024-09-06:
Hi, I would also be grateful if you could post your full yaml. I have just purchased a 7" version and having problems integrating your code that you mentioned works.
With thanks
Andy

AndyDann, 2024-09-06:
After some "playing" I have now been able to get the 7" CrowPanel board (v3) working with LVGL

Here's my code for reference... you'll need to change the images of course from my code.

Hope this helps someone and thanks for the above code.

`esphome:
  name: "crowpanel7"
  friendly_name: CrowPanel 7
  platformio_options:
    build_flags: "-DBOARD_HAS_PSRAM"
    board_build.esp-idf.memory_type: qio_opi
    board_build.flash_mode: dio
  #  board_upload.maximum_ram_size: 524288

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: esp-idf
    #version: recommended
    sdkconfig_options:
      CONFIG_ESP32S3_DEFAULT_CPU_FREQ_240: y
      CONFIG_ESP32S3_DATA_CACHE_64KB: y
      CONFIG_SPIRAM_FETCH_INSTRUCTIONS: y
      CONFIG_SPIRAM_RODATA: y

psram:
  mode: octal
  speed: 80MHz
  
# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "AIdWswWRR5YSvp7Itw7qPNe5WmkBqWoszqfjWgEuCNI="

ota:
  - platform: esphome

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

external_components:
  - source: github://pr#7184
    refresh: 10min
    components: [lvgl]

# Define a PWM output on the ESP32
output:
  - platform: ledc
    pin: 2
    frequency: 1220
    id: gpio_backlight_pwm

light:
  - platform: monochromatic # Define a monochromatic, dimmable light for the backlight
    output: gpio_backlight_pwm
    name: CrowPanel7 Backlight
    id: back_light
    restore_mode: ALWAYS_ON

i2c:
  sda: 19
  scl: 20
  scan: true

display:
  - platform: rpi_dpi_rgb
    id: main_display
    data_pins:
      red:
        - 14        #r1
        - 21        #r2
        - 47        #r3
        - 48        #r4
        - 45        #r5
      green:
        - 9         #g0
        - 46        #g1
        - 3         #g2
        - 8         #g3
        - 16        #g4
        - 1         #g5
      blue:
        - 15        #b1
        - 7         #b2
        - 6         #b3
        - 5         #b4
        - 4         #b5
    de_pin: 41
    hsync_pin: 39
    vsync_pin: 40
    pclk_pin: 0

    hsync_front_porch: 40
    hsync_pulse_width: 48
    hsync_back_porch: 13

    vsync_front_porch: 1
    vsync_pulse_width: 31
    vsync_back_porch: 13

    pclk_inverted: true
    pclk_frequency: 8500000

    color_order: RGB

    auto_clear_enabled: false # takes 2.8 seconds to clear the display
    update_interval: 5s

    dimensions:
      width: 800
      height: 480

image:

  - file: images/Solar_Battery.png
    id: solarbattery
    resize: 60x60
    type: RGB565   
    use_transparency: true

lvgl:
  log_level: INFO
  color_depth: 16
  bg_color: 0x000020
  border_width: 0
  outline_width: 0
  shadow_width: 0
  text_font: unscii_16
  align: center
  style_definitions:
    - id: font_style
      text_font: MONTSERRAT_32
      align: center
      text_color: 0xFFFFFF
      bg_opa: TRANSP
      bg_color: 0
      radius: 3
      pad_all: 2
 
  widgets:

    - button:
        x: 150
        y: 200
        width: 70
        height: 28
        id: btn_low
        checkable: false
        widgets:
          - label:
              align: center
              text_font: MONTSERRAT_18
              text: "Low"
    - image:
        x: 400
        y: 240
        id: weather_icon_today
        src: solarbattery

`

heibertelf, 2025-01-17:
>     hsync_front_porch: 40
>     hsync_pulse_width: 48
>     hsync_back_porch: 13
> 
>     vsync_front_porch: 1
>     vsync_pulse_width: 31
>     vsync_back_porch: 13

I found EK9716BD datasheet, and it has different values:
[image]
so maybe for some displays this configuration could be more correct:
```
    hsync_front_porch: 40
    hsync_pulse_width: 48
    hsync_back_porch:  40

    vsync_front_porch: 13
    vsync_pulse_width: 1
    vsync_back_porch:  31

```