esphome/feature-requests #2748: Support for Pylontech H2 Force battery

mamo-nick, 2024-06-06:
**Describe the problem you have/What new integration you would like**

Adding H2 Force battery support to ESP Home Pylontech integration.

**Please describe your use case for this integration and alternatives you've tried:**

Monitoring of individual batteries, number of charge cycles and other values (temperature, current, voltage​​... Pylontech H2 Force battery

**Additional context**

RFDarter, 2024-06-27:
Can you provide some uart loogs to see how the communication differ from other pylontech batteries?

mamo-nick, 2024-07-07:
I have no uart conection, so I can not provide logs.
Do you think I should connect my bms like if I have US2000C for example?