esphome/feature-requests #2752: climate platform for using modbus registers

krismaddens, 2024-06-08:
**Describe the problem you have/What new integration you would like**
A climate platform where you can link any sensor as room sensor and any number as target temperature so that this can be linked  to a modbus addresses where the assigned modbusregisters are updated like any other climate entity. 

**Please describe your use case for this integration and alternatives you've tried:**
 the problem is that it is now not possible when using the thermostate platform to link a modbus_controller number (or any random number) to the set temperature of a climate in a way that the number is automaticaly updated when using the climate entity in HA. in this case there are no other elements to be controlled but the measured temperature, the set temperature ans the mode.  

**Additional context**
 there are other platforms where there is no actions needed like the Tuya or Midea platform where there is no actual output to be controlled but only to pas on the sensor, set and mode values. so if that can be modified for use with any sensor, number and sellect a lot of ideas can be realized

nagyrobi, 2024-06-08:
Why do you make two feature requests for essentially the same thing?

krismaddens, 2024-06-08:
because trere are 2 ways to solve this, or the modbus_controler gets a climate comonent or the climate gets a platform to combine sensor, number and sellect. and mayby there are members in favor of working on the modbus component sollution or  there are members that find working on the climate platform a beter sollution.

krismaddens, 2024-06-08:
or we get sollutions on both and the way to a perfect esphome is yet again shorter :)

krismaddens, 2024-06-08:
and there is also a 3dr way to solve this, if in HA we can create a climate entity with whatever entity as sensor, target, mode. that would solve also a lot of things. The thing is if you are like me, a plain belgian farmer that is stil in the proces of understanding "lambda" the only thing i can do is search for sollutions from others or bring on the question and sugestions.

nagyrobi, 2024-06-08:
There are many ways to solve your request. Either way, the goal is that you have a modbus hvac device, which you'd like to appear as a climate in ESPHome. Devs will have a stronger opinion on how is technically more appropriate to accomplish this.

Can you share what exact modbus devices are involved in your case?

krismaddens, 2024-06-08:
The device is a pool control device with pH, orp,temperature controller and flow, canister level,... input and otput relays.
it has a TTL serial modbus comunication. The goal is to create a gateway modbus/ esphome so that all the Yaml hasle and automations that is otherwise needed in HA are handled on a esp32 board with esphome. and that works fine. al what is not working fine is the temerature control. the actual temperature control is done on the device itself. now the target temerature is a number in esphome, works perfect if you are a fan of a slider or other numeric input in HA but it is not possible to create a climate entity that can be used in a thermostat card in a way that everything is bidirectional updated.

krismaddens, 2024-06-08:
a modbus_controller climate component where we can assign the modbus addresses for sensor, target and mode  or a climate platform where any sensor id, number id and sellect id can be used to create a climate entity is what i need. That way the target temperature number id, temperature sensor ID, mode sellect id can be used internaly in the esphome for further automations.