esphome/feature-requests #2755: Modbus binary sensor with inverted property

vespadj, 2024-06-10:
**Describe the problem you have/What new integration you would like**

Modbus. Read the state of a Normal Close switch.
ON means ok, OFF means not ok.
But setting entity_category as diagnostic, signal must be inverted.
`inverterd: true` not allowed for `binary_sensor` \ `platform: modbus_controller`

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

nagyrobi, 2024-06-10:
Try the [invert](https://deploy-preview-3927--esphome.netlify.app/components/binary_sensor/#invert) filter.

vespadj, 2024-06-10:
`filters: - invert:` works fine. Thanks

> Try the [invert filter](https://esphome.io/components/binary_sensor/#binary-sensor-filters).

I suggest for analogy, a [property like in GPIO](https://esphome.io/components/binary_sensor/gpio#inverting-values), but feel free to close.
```
binary_sensor:
  - platform: gpio
    pin:
      number: D2
      inverted: true
```

closed 2024-06-10 by nagyrobi