esphome/feature-requests #2759: Document environment variables

gabest11, 2024-06-12:
**Describe the problem you have/What new integration you would like**

The official documentation barely mentions any info on environment variables used by esphome. I went through the code and collected them, it would be nice if someone, preferable the one who added it, could write a short description for each.

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

### ESPHOME_BUILD_PATH

https://esphome.io/components/esphome.html

build_path (Optional, string): Customize where ESPHome will store the build files for your node. By default, ESPHome puts the PlatformIO project it uses to build the firmware in the .esphome/build/<NODE> (or into path from ESPHOME_BUILD_PATH environment variable if specified) directory, but you can customize this behavior using this option. Official docker image automatically use /build folder as default one in case it is mounted to it.

### ESPHOME_TRUSTED_DOMAINS

https://github.com/esphome/esphome/pull/6845

A comma separated list of allowed domains. 

If the ESPHOME_TRUSTED_DOMAINS is set, the overriden check_origin compares it with the http origin header, otherwise it calls the parent and works just like before.

### USERNAME, PASSWORD

UI authentication

### ESPHOME_DASHBOARD_RELATIVE_URL
### ESPHOME_DASHBOARD_USE_PING
### ESPHOME_STREAMER_MODE
### ESPHOME_DASHBOARD_USE_MQTT
### DISABLE_HA_AUTHENTICATION

(dashboard/settings.py, happened during dashboard.py split, cannot find origin)

https://github.com/esphome/esphome/pull/461

???

### ESPHOME_IS_HA_ADDON

(is_ha_addon: helpers.py, core/__init__.py)

???

### ESPHOME_DASHBOARD_MQTT_USERNAME
### ESPHOME_DASHBOARD_MQTT_PASSWORD
### ESPHOME_DASHBOARD_MQTT_BROKER
### ESPHOME_DASHBOARD_MQTT_PORT

(mqtt.py)

???

### SUPERVISOR_TOKEN

(dashboard/web_server.py)

???

### ESPHOME_QUICKWIZARD (ENV_NOGITIGNORE)
### ESPHOME_NOGITIGNORE (ENV_DEV)

(wizard.py)
(dashboard/web_server.py)

???

### ESPHOME_BUILD_PATH

(core/config.py)

???

### ESPHOME_DATA_DIR

(core/__init__.py)
(is_ha_addon ? /data : ESPHOME_DATA_DIR ? ESPHOME_DATA_DIR : .esphome)

???

crazytimmy, 2025-01-03:
Unfortunate there's no updates on this.

@gabest11 did you notice if they're using secrets in the env variables? I'd like to pass secrets for the username & password for the dashboard auth specifically.

mmathau, 2025-07-01:
> Unfortunate there's no updates on this.
> 
> [@gabest11](https://github.com/gabest11) did you notice if they're using secrets in the env variables? I'd like to pass secrets for the username & password for the dashboard auth specifically.

I just found this issue looking for docker/swarm secrets support for `USERNAME` and `PASSWORD` but it seems like there is no way at the moment without using a custom `entrypoint.sh`.