esphome/feature-requests #2761: WCH32 Processor Support

bugeaud, 2024-06-14:
**Describe the problem you have/What new integration you would like**

I would like WCH32 support on ESPHome.

At this time various platform such as Tensilica/Xtensa (ESP8266, ESP32, ESP32-S2) or RISC-V (ESP32-C3) are supported thru ESI IDF.

WCH32 is a broad range of MCU that are RISC-V compatible. From very light 20c$ CH32V003 a pop, to around 1.3$ a pop for a more complete CH32V305 that even includes Ethernet MAC and Ethernet 10M PHY transceivers.

I would expect that we would be able to use board using WCH32 as alternative to existing ESP platforms.

**Please describe your use case for this integration and alternatives you've tried:**

Having WCH32 supported would be interresting to ensure broader esphome support from various manufacturer. ESP offers quite complementary wire and wireless solutions to what WCH is proposing. ESP is more on the wireless side whereas WCH is more on the wire side.

We could get board using WCH32 and get them working on esphome.

**Additional context**

This will require an in depth understanding of the dev stack / tool chain compatibility between ESP and WCH and maybe a building of several adapters or go-between shims.

nagyrobi, 2024-06-14:
I assume the first step should be to get a handful of dev boards with such MCUs to the devs, testers.

randybb, 2024-06-14:
Just check https://github.com/esphome/esphome/pull/6075 (nrf52 for super low power BLE/zigbee sensors) how much fun it is :)
Yes, it is cheap, but I don't think worth as esp32 is not crazy expensive.

Anyway, they have a really nice offline programmer, that can be used as a dev board as well https://www.youtube.com/watch?v=Qnt_43PZcyo

nagyrobi, 2024-06-14:
The very first break of ice for ESP32 wasn't easy either. When only 8266 existed and everyone was just wowing for ESP32 which looked like a completely different kind of animal.

bugeaud, 2024-06-14:
@nagyrobi 

Example of such a dev board with onboard Ethernet https://fr.aliexpress.com/item/1005004449629983.html

I've been trying to dig esphome src for some while, but I am stil not clear how far is ESPhome tied to esp-idf. Because, my guess is that this is the key. The more ESPhome is not tied to esp-idf the "easier" a port would get for each module/feature/component. I anticipate a port would be very different and more or less easy depending on the components.

nagyrobi, 2024-06-14:
Check out the [LibreTiny](https://github.com/libretiny-eu) project approach they did case of some Tuya chips

Hedda, 2024-06-16:
> I've been trying to dig esphome src for some while, but I am stil not clear how far is ESPhome tied to esp-idf. Because, my guess is that this is the key. The more ESPhome is not tied to esp-idf the [...]

I believe that all ESPHome platforms supported uses the Arduino framework by default, (not ESP-IDF framework by default though that is supported as an alternative base framework for ESP32 chips, however ESP-IDF is for example for used for the ESP8266 platform or the LibreTiny platform ). 

https://esphome.io/components/esp32.html

https://esphome.io/components/esp8266.html

https://esphome.io/components/libretiny.html

Regardless, ESPHome platform is also depending on libraries also being supported by PlatformIO for build and compilation.

https://esphome.io/components/esphome.html#platformio-options

https://github.com/esphome/esphome/blob/dev/platformio.ini