esphome/feature-requests #2763: support for multi channel SPI sensors using cd74hc4067 or similar muxes

gkasprow, 2024-06-16:
**Describe the problem you have/What new integration you would like**

I want to connect many SPI energy/temperature meter ICs to ESP32 with Ethernet. I don't have that many pins.
I use a multiplexer like cd74hc4067  to get many CS signals in such cases.
With such a chip, I need four pins for SPI and four for CS mux, and I can talk to 16 SPI devices.
I can even spare one pin and treat output 0 of cd74hc4067  as none selected and connect 15 devices to remaining outputs 1..14.

**Please describe your use case for this integration and alternatives you've tried:**

I want to connect 15 or 16 [ATM90E26](https://esphome.io/components/sensor/atm90e26) sensors to ESP32 having Ethernet PHY chip which consumes most of it's pins.

**Additional context**

I'm willing to design an open hardware platform for such energy/temperature meters.
I authored over 150 OH designs (gkasprow, elhep, sinara-hw repositories)

sbrk-r, 2025-01-25:
I too have the same problem, ethernet consumes a lot of pins and individual CS are such a waste.
  
I think 74HC139 (2:4), '138 (3:8), and '154 (4:16) are more suitable, but the software side would be the same anyway. Convert n GPIOs into 2^n CS# lines.

gkasprow, 2025-01-29:
My colleague implemented this feature already
https://github.com/wizath/esphome/tree/dev

ssieb, 2025-01-29:
A 4067 is an analog chip and it's only input, so not suitable at all.  Why not just use one of the i2c gpio expanders?  Then you only need 2 pins for as many CS pins as you could want.

@gkasprow I don't see anything there.  That's just an out-of-date clone of esphome.  There aren't even any branches.

gkasprow, 2025-01-29:
you can't use expander. 
first, SPI would be damn slow,
second, it's not supported by ESPHOME to control CS lines
4067 is fine, it doesn't care if signal is analog or digital. Just add pullups on it's outputs and connect input to GND.
I asked my coleague where is the repo

ssieb, 2025-01-29:
You're not using the expander for the SPI, just the CS pin and yes, that's supported.

ssieb, 2025-01-29:
```yaml
    cs_pin:
      mcp23xxx: mcp23008_hub
      number: 0
```