esphome/feature-requests #2765: LTR390 different gain and resolution for ALS and UV

tgrosz, 2024-06-17:
**Describe the problem you have/What new integration you would like**

The LTR390 uses one gain and resolution setting for both ALS and UV measurements. Outdoors it's not the best solution.

**Please describe your use case for this integration and alternatives you've tried:**

I'd like to use smaller gain and resolution for ALS measurement outdoors, so that both ALS and UV measurements would be usable

**Additional context**

In arduino lib example code, they are used with different values
https://registry.platformio.org/libraries/levkovigor/LTR390/examples/ltr390/ltr390.ino

latonita, 2024-07-01:
@tgrosz ltr390 doesnt support different gains for als/uv. That library is just switching gain and mode and then read data.
I think I can update the component to do similar approach.

latonita, 2024-07-01:
@tgrosz try it.

Gain can be set as usual - same value for ALS and UV (
```yaml
    gain: X18
```
or granular
```yaml
    gain:
      ambient_light: X9
      uv: X3
```

```yaml
external_components:
 - source: github://pr#7026
   components: [ltr390]

sensor:
  - platform: ltr390
    uv_index:
      name: "UV Index"
    uv:
      name: "UV Sensor Counts"
    light:
      name: "Light"
    ambient_light:
      name: "Light Sensor Counts"
    gain:
      ambient_light: X9
      uv: X3
```

tgrosz, 2024-07-01:
Hi @latonita , thanks - as I see it's working OK.
My idea was the same - to set the gain before reading the value

latonita, 2024-07-02:
@tgrosz please test separate resolution. 
similarly:
```yaml
resolution:
  abient_light: 20
  uv: 13
```

tgrosz, 2024-07-02:
@latonita I get error message
`  Unknown value '{'ambient_light': 18, 'uv': 20}', valid options are '20', '19', '18', '17', '16', '13'.
  resolution: 
    ambient_light: 18
    uv: 20`
I cleared build files, what else should I do?

latonita, 2024-07-02:
Share your config please. Ltr390 part

latonita, 2024-07-02:
I have an idea - check if you have white spaces like this 'resolution: ', remove if they exist. I will check this also

tgrosz, 2024-07-02:
Tried deleting indentation and adding spaces manually, did not help
`  - platform: ltr390
#    gain: X18
    gain:
      ambient_light: X3
      uv: X18
#    resolution: 20
    resolution:
      ambient_light: 18
      uv: 20
    id: ltr390_
    update_interval: 60s
    uv_index:
      name: "UV Index"
      id: ltr390_uvi
      unit_of_measurement: "UVI"
    uv:
      name: "UV"
      internal: true
    light:
      name: "Light"
      id: ltr390_light
    ambient_light:
      name: "Ambient Light"
      internal: true`

latonita, 2024-07-03:
This is strange.
Please send full yaml

tgrosz, 2024-07-03:
Strange, now it compiles without any change. Maybe some cache issue.
So, it works, Thanks

closed 2024-07-12 by jesserockz