esphome/feature-requests #2767: Add WhatsApp notifications

mgrouch, 2024-06-18:
Allow sensor to send WhatsApp notifications.

**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

example of code sending WhatsApp messages from esp32

https://github.com/bareboat-necessities/bbn-m5atomS3-lite/tree/main/bbn_m5atomS3_lite_whatsapp

**Additional context**

nielsnl68, 2024-06-18:
Is this not something that you want to have in HA?

mgrouch, 2024-06-18:
I really like that I can use EspHome sensors without home assistant

RFDarter, 2024-06-27:
It looks like it is a simple http request.
So it seems this is allready possible without too much work.

WebSpider, 2024-08-07:
Looking at https://github.com/bareboat-necessities/bbn-m5atomS3-lite/blob/58661d2885d39773ebd345e921bf973877ae3451/bbn_m5atomS3_lite_whatsapp/bbn_m5atomS3_lite_whatsapp.ino#L89 it is indeed a simple HTTP request.

This request can probably be done with the http integration 
here: https://esphome.io/components/http_request