esphome/feature-requests #2780: Time Based Cover with Tilt Function  by klaudiusz223: - manual_control / has_built_in_endstop  Functions

sleepymaxx, 2024-06-23:
**Describe the problem you have/What new integration you would like**

Integration of manual_control / has_built_in_endstop Functions From time_based to time_based_tilt component by Klaudiusz223

**Please describe your use case for this integration and alternatives you've tried:**

covers not closing/not opening/ wrong tilt .....

**Additional context**

@ all Readers / Inventors / Userers thanks for your great Work!

@ klaudiusz223: Thankfully i installed your esphome@tilt-cover and its working like a charm.. only with minor faults: 
Code down below my explanation of my problem: 
First: i installed your code onto a ESP8266 D2 mini device and its working like a charm   -GREAT THANKS FOR THIS

[https://github.com/klaudiusz223/esphome-docs/blob/tilt-cover/components/cover/time_based_tilt.rst](url)

Only in my case there are additional manual Controls on which all of the covers can be steered maually - so the assumed state of the covers are in all stations not really right - i reworked this by the automations which are steering my covers form time to time this is getting me some issues (covers not closing/not opening/ wrong tilt .....) - but i read in your documentation of your Time Based integration 
[https://github.com/klaudiusz223/esphome-docs/blob/tilt-cover/components/cover/time_based.rst](url) there is a option to integrate this into the code of the esp by itself - YOUR "manual_control" boolean but unfortunately it is not working in the esphome@tilt-cover integration. 
Also your has_built_in_endstop Function from the above mentioned integraton will come in handy if added to your time_based_tilt component.

1. post script: sorry for bad english.... school was very much back in time 
2. post script: if i lacked using the right code or implementation sorry for this... im already a learer in this things and don´t really understand coding or anything like this - all im doing is reading and much copying... 
3. post script: thanks in advance...

`logger: 

binary_sensor:

external_components:
  - source: github://klaudiusz223/esphome@tilt-cover
    components: [ time_based_tilt ]

switch:
  - platform: gpio
    id: "blind_01_relay_down"
    disabled_by_default: true
    pin:
      number: GPIO4
      mode:
        output: true
      inverted: true

  - platform: gpio
    id: "blind_01_relay_up"
    disabled_by_default: true
    pin:
      number: GPIO3
      mode:
        output: true
      inverted: true

  - platform: gpio
    id: "blind_02_relay_down"
    disabled_by_default: true
    pin:
      number: GPIO12
      mode:
        output: true
      inverted: true

  - platform: gpio
    id: "blind_02_relay_up"
    disabled_by_default: true
    pin:
      number: GPIO5
      mode:
        output: true
      inverted: true

cover:
  - id: cover_01
    platform: time_based_tilt
    name: "Jalousie Sued"
    open_duration: 49700 ms
    close_duration: 48700 ms
    inertia_open_time: 105 ms
    tilt_open_duration: 900 ms
    inertia_close_time:  105 ms
    tilt_close_duration: 900 ms
    interlock_wait_time: 105 ms
    recalibration_time: 1500ms
    assumed_state: true

    close_action:
      - switch.turn_on: blind_01_relay_down
    open_action:
      - switch.turn_on: blind_01_relay_up   
    stop_action:
      - switch.turn_on: blind_01_relay_up
      - switch.turn_on: blind_01_relay_down	 
      - delay: 105 ms
      - switch.turn_off: blind_01_relay_up
      - switch.turn_off: blind_01_relay_down

  - id: cover_02
    platform: time_based_tilt
    name: "Jalousie West"
    open_duration: 49700 ms
    close_duration: 48700 ms
    inertia_open_time: 105 ms
    tilt_open_duration: 900 ms
    inertia_close_time: 105 ms
    tilt_close_duration: 900 ms
    interlock_wait_time: 105 ms
    recalibration_time: 1500ms
    assumed_state: true

    close_action:
      - switch.turn_on: blind_02_relay_down
    open_action:
      - switch.turn_on: blind_02_relay_up
    stop_action:
      - switch.turn_on: blind_02_relay_up
      - switch.turn_on: blind_02_relay_down	 
      - delay: 105 ms
      - switch.turn_off: blind_02_relay_up
      - switch.turn_off: blind_02_relay_down     `

klaudiusz223, 2024-11-24:
Hi @sleepymaxx 
Please try the latest version of my implementation esphome/esphome#6146.
There is no need to set an additional parameter such as manual_control.
Just make sure that you set the recalibration time and that it is long enough, i.e. equal to or slightly longer than the open or close time.
Please note that the configuration has changed slightly and is not compatible with the previous version.
In the latest implementation, blinds that are already fully open or closed will enter a recalibration state.

sleepymaxx, 2024-11-25:
Somehow i got a really bad problem here: 

``INFO ESPHome 2024.11.1
INFO Reading configuration /config/esphome/rollaeden-jalousien-esp.yaml...
ERROR Unexpected exception while reading configuration:
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/esphome/esphome/__main__.py", line 1036, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1014, in run_esphome
    config = read_config(dict(args.substitution) if args.substitution else {})
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 1091, in read_config
    res = load_config(command_line_substitutions)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 945, in load_config
    return _load_config(command_line_substitutions)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 935, in _load_config
    return validate_config(config, command_line_substitutions)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 810, in validate_config
    do_external_components_pass(config)
  File "/esphome/esphome/components/external_components/__init__.py", line 116, in do_external_components_pass
    _process_single_config(c)
  File "/esphome/esphome/components/external_components/__init__.py", line 77, in _process_single_config
    components_dir = _process_git_config(
                     ^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/components/external_components/__init__.py", line 43, in _process_git_config
    repo_dir, _ = git.clone_or_update(
                  ^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/git.py", line 90, in clone_or_update
    age = datetime.now() - datetime.fromtimestamp(file_timestamp.stat().st_mtime)
                                                  ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/pathlib.py", line 1014, in stat
    return os.stat(self, follow_symlinks=follow_symlinks)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: '/data/external_components/1e020dc3/.git/HEAD'

and can´t test your integration ... honestly im very sad about this ... i really apreciate your work and this will be the best way to push it further ... but i got errors (only in this esp yaml.... ) sorry about this --- will work as much as possible to get my problem solved! thank you anyway!!

sleepymaxx, 2024-11-25:
unfortunately i have really no idea how to fix this ..

klaudiusz223, 2024-11-26:
Hi @sleepymaxx 

Try "Clean build files" option
Change the configuration to
```
external_components:
  - source: github://pr#6146
    components: [ time_based_tilt ]
```

but note that the configuration has changed. The configuration description is esphome/esphome#6146.

sleepymaxx, 2024-12-04:
Again you saved my day.... all is working fine now. 
best wishes for ongoing Christmas

closed 2024-12-04 by sleepymaxx