esphome/feature-requests #2781: Add support for KP18058 / 68 "I2C" variant peripheral chip

Nuetron06, 2024-06-24:
**Describe the problem you have/What new integration you would like**

The KP18058 / 68 "I2C" chip is used in some RGBCW Tuya lightbulbs paired with the CB2L chip to control the 5 LED channels of the light bulb. The particular one I have is branded XTreme and purchased from a Five Below store. 

**Please describe your use case for this integration and alternatives you've tried:**

Currently the chip is not supported by ESPHome and does not have a standard implementation of I2C.

**Additional context**

Martin has posted links [here](https://github.com/esphome/esphome/discussions/5418), but unfortunately the links do not match the text. See my post right after his for the corrected links.

LukasMojzis, 2026-04-26:
Whatever happened to this component?

oxynatOr, 2026-10-06:
https://github.com/oxynatOr/esphome-kp18058

i still wait for my bulb :/ so i couldn't test it.