esphome/feature-requests #2782: Support ESL (electronic shelf label) BLE displays

Misiu, 2024-06-24:
**Describe the problem you have/What new integration you would like**

There are projects that allow users to use ESL'a as a portable displays and with long battery life, they are ideal for dashboards.
Recently some BLE based tags showed up, like this one: https://pl.aliexpress.com/item/1005002399342939.html
We can pick 2.1, 2.9, 4.2, 7.5 and 10.2 inch displays.

Projects like https://github.com/OpenEPaperLink allows using ESP32-S3 as a gateway to communicate with the display.
This requires additional hardware (another ESP32) and a dedicated integration to be able to display anything from HA.

With dedicated support we could use ESPHome draw API to create content and send it to the ELS displays.

**Please describe your use case for this integration and alternatives you've tried:**

I have currently two displays in my house, both are used to display waster collection data, but I already plan to add additional ESL's (4,2 inch) to display weather info and some NAS statistics.

This is my design so far:
[image]

**Additional context**

See example displays for inspirations: https://github.com/OpenEPaperLink/Home_Assistant_Integration/wiki/1.54%22%2C-2.9%22-and-4.2%22-Weather-display-using-met.no%2C-Moon-phase%2C-Gotham-Round-font

nagyrobi, 2024-06-24:
How do these work? You actually just send over the rendered bitmap in data chunks to the display?

Misiu, 2024-06-24:
@nagyrobi Sadly I'm not that technical.
Currently, I use the integration to create the image, the integration sends it to the AP (running OpenEpaperLink) and the image is sent to the tag via Bluetooth.
In the mentioned issue there are links to the ble_filter and able writer:
https://github.com/OpenEPaperLink/OpenEPaperLink/blob/master/ESP32_AP-Flasher/src/ble_filter.cpp
https://github.com/OpenEPaperLink/OpenEPaperLink/blob/master/ESP32_AP-Flasher/src/ble_writer.cpp

nagyrobi, 2024-06-24:
> create the image, the integration sends it to the AP (running OpenEpaperLink) and the image is sent to the tag via Bluetooth.

https://github.com/OpenEPaperLink/OpenEPaperLink/wiki/Building-an-Access-Point#why-do-i-need-an-esp32-s3-and-a-esp32-c6

So I assume that means we need a display platfrorm too, which could pass the image rendered to the AP side of things, which is an ESP32-C6 running in IEEE 801.15.4 mode continuously, talking to the tags.

Misiu, 2024-06-24:
@nagyrobi my FR was about BLE based Tags, like this one: https://pl.aliexpress.com/item/1005002399342939.html
I have a single ESP-S3 flashed with OpenEpaperLink (they have a standalone Bluetooth-only AP).
It works perfectly.
Ideally, I'd like to use a single ESP-S3 to run ESPHome, and ESPHome should send the image directly to the TAG.

nagyrobi, 2024-06-24:
Bluetooth and WiFi coexistence is a difficult thing, that's what the OpenEPaperLink wiki page also states.
If you look at their [Yellow AP](https://github.com/OpenEPaperLink/OpenEPaperLink/tree/master/Hardware/Yellow%20AP%20by%20ATC1441), the proper solution usable by anyone would be as described above, have the ESP32-C6 run OpenEPaperLink firmware, and have the ESP32-S3 run ESPHome firmware.

Misiu, 2024-06-24:
If you want to connect to other types of tags then yes, you need additional ESP.
But there is BLE only AP - https://install.openepaperlink.de/
I asked about that just weeks ago - https://github.com/OpenEPaperLink/OpenEPaperLink/issues/307#issuecomment-2156772501
You need ESP32-S3 N16R8, you flash the BLE-only firmware and you have an ESP flashed with firmware that allows connection to BLE-only tags.
So if that firmware can connect to the BLE tag and my wifi I think the same thing could be done in ESPHome - generate the image data and send it to the tag.

Ares4570, 2025-04-15:
I'd be interested in this as well (again, for BLE tags/displays) - I have ESP32S3s scattered throughout my house, and being able to utilize existing devices vs needing to buy/flash/hide new hardware is always ideal.