esphome/feature-requests #2783: Template climate platform

nagyrobi, 2024-06-25:
**Describe the problem you have/What new integration you would like**
There are many cases where directly unsupported HVAC devices could be added to ESPHome if a Template climate platform would exist.
- for unsupported IR devices lambdas could be used to call directly various library functions (see Hagakurje's example below)
- standard ESPHome triggers and actions could be used to call ModBUS or even Remote Transmitter stuff
- some simpler devices could be controlled with switches etc.

**Please describe your use case for this integration and alternatives you've tried:**
Here somebody implements a `custom` climate to control an unsupported `gree` model by calling in directly `IRremoteESP8266`: https://github.com/Hagakurje/ESPHome_Gree_AC
(it is known that `custom` platforms are to be deprecated/removed)

Here someone is sending captured IR commands in RAW or other likesome protocols using `remote_transmitter` platform, and integrates the climate functionality as template fan: https://github.com/esphome/feature-requests/issues/2240

Here someone has a HVAC device accessible via ModBUS would like to implement it as a climate: https://github.com/esphome/feature-requests/issues/2752

**Additional context**
The concept behind Tasmota's IRHVAC is nice: https://tasmota.github.io/docs/Tasmota-IR/#sending-irhvac-commands
It's basically a wrapper around `IRremoteESP8266` where you just select a protocol and the commands you'd like to send.

A kind of template climate component would be very flexible in ESPHome, where anyone could mix and match the features supported by `IRremoteESP8266` and his device. Eg. some devices only have vertical swing, then let the component not expose the horizontal swing. Advanced functions could be called with template switches.

rob-deutsch, 2024-06-25:
I don't have any opinion on this. I am just the codeowner for the `heatpumpir` integration.

If you're wanting to do something novel then I'd recommend reading the PR discussion for the `heatpumpir` integration where the pros/cons were discussed.

https://github.com/esphome/esphome/pull/1343

CiniBN, 2024-09-26:
Hello!
Will there be something from this platform?
I am surprised that it does not exist yet.  There are countless devices that communicate on Modbus and this platform would be very useful for that. I currently have a heat pump that I integrated with ESP32 on Modbus with select and switch components, but it's not the real thing.  Robi! Te tudsz erről valami biztatót?

polyfloyd, 2024-12-29:
I have a modbus based heatpump that I would like to integrate with a climate template-like solution. I have been working on an implementation for this for the past few days and am planning on upstreaming it when I deem it good enough.

This is how my configuration now looks like:
```
climate:
  - platform: template_climate
    name: "LG Therma"
    target_temperature_id: true_setpoint
    current_temperature_id: climate_current_temperature
    mode_id: climate_mode
    action_id: climate_action
    supported_modes:
      - "OFF"
      - "COOL"
      - "HEAT"
      - "AUTO"
    temperature_step: 0.1C
```

This snippet references number and selects for the respective states it needs. The target_temperature_id references a number with the modbus_controller platform. climate_current_temperature is a sensor and action is a text_sensor.

The mode is a template select to handle the "OFF" state, which has to be routed to a distinct modbus address:

```
select:
  - platform: template
    id: climate_mode
    lambda: !lambda |-
        return id(lg_therma_active).state
             ? id(lg_therma_mode).state
             : "OFF";
    set_action:
        - lambda: !lambda |-
            if (x == "OFF") {
              id(lg_therma_active).state = false;
            } else {
              id(lg_therma_mode).state = x;
              id(lg_therma_active).state = true;
            }
    options:
      - "OFF"
      - "COOL"
      - "HEAT"
      - "AUTO"

  - platform: modbus_controller
    modbus_controller_id: lg
    id: lg_therma_mode
    name: Operation Mode
    icon: mdi:sun-snowflake-variant
    address: 0
    value_type: U_WORD
    optionsmap:
      "COOL": 0
      "HEAT": 4
      "AUTO": 5

switch:
  - platform: modbus_controller
    id: lg_therma_active
    name: "Active"
    icon: mdi:power
    modbus_controller_id: lg
    register_type: coil
    address: 0

```

I considered having this template climate component to have actions available to be set, but I think it is better to delegate such implementation details to other template components and have this component just be the collective state.

LukasGrebe, 2025-05-05:
+1 looking to integrate my modbus heatpump as a native climate device for its UI in HomeAssistant.
A generic template climate would allow forwarding set temp, current temp/action, mode ect.

closed 2026-09-06 by ssieb