esphome/feature-requests #2784: jsn_sr04t component: AJ_SR04M compatibility mode

soeffi, 2024-06-25:
**Describe the problem you have/What new integration you would like**
The AJ_SR04M ultarsonic module is very similar to the JSN SR04T and has similar operation modes.

There is only one differences in the protocol, which is the checksum calculation that does not include the first byte.

Instead of adding a new component, would it be possible to add a configuration variable (e.g. `ajsq04m_mode`, which defaults to false), which, if set to true, changes the checksum calculation from

`uint8_t checksum = this->buffer_[0] + this->buffer_[1] + this->buffer_[2];`
to
`uint8_t checksum = this->buffer_[1] + this->buffer_[2];`
?  

**Please describe your use case for this integration and alternatives you've tried:**

see above

**Additional context**

see above

soeffi, 2024-06-30:
@Mafus1, you are the original author of this component. Would you be willing to support and do this change?

The needed code change is probably relatively small for someone who knows where to look for...

Mafus1, 2024-07-04:
@soeffi, I'm willing to look into it. Do you own the appropriate hardware and are able to test necessary changes?

soeffi, 2024-07-04:
Great news, thanks a lot!

I do own an AJ_SR04M sensor and I'd use it connected to a Wemos D1 Mini for testing. I might need some guidance in how to best do testing. Currently I'm using ESPhome via the web frontend hosted in a docker container.

If I understand correctly, according to https://esphome.io/components/external_components, the changes could be developed in a pull request, which I can refer directly out of the yaml configuration?

If that's not possible, I can set up a dev environment.

As a first step, I guess I'd like to check whether the simple change of the checksum calculation makes things work as desired with the sensor. If that's the case, in a second step it probably makes sense to add a configuration option that enables the different checksum calculation

soeffi, 2024-07-04:
Ok, this external components thing is actually really nice. I just created a fork to test the alternative checksum calculation for AJ_SR04M and that works as expected.

Now all that's needed is a config option (boolean?) which enables AJ_SR04M mode. Currently this is beyond my esphome and python skills. Will try to find a good source to learn how to get that into sensor.py...

closed 2024-07-11 by jesserockz