esphome/feature-requests #2786: Add support for Trotec Pac 2010 E A/C

marius-arch, 2024-06-27:
**Describe the problem you have/What new integration you would like**
I would like to control my A/C via infrared using the ESP Home IR Remote Climate. Currently my  Airconditioner is not supported.

**Please describe your use case for this integration and alternatives you've tried:**
Support for the Trotec PAC 2010 E would allow me integrating it into my HomeAssistant.

**Additional context**

nagyrobi, 2024-06-27:
Did you try the `zhlt01` climate or protocol, or the `zhjg01` protocol from [IR Remote Climate](https://esphome.io/components/climate/climate_ir)? These are remotes used with many locally branded air conditioners.

marius-arch, 2024-06-27:
Ok thank's, i will give that a try

nagyrobi, 2024-06-27:
To try `zhjg01` you need this, until it's merged and released:
```
external_components:
  - source: github://pr#6996
    components: 
      - heatpumpir
    refresh: 5min
```

marius-arch, 2024-07-14:
Sadly, neither zhlt01 nor zhjg01 worked for me. Do you have any other suggestions? I saw that IRremoteESP8266 supports A/C devices from trotec

nagyrobi, 2024-07-14:
No other posdibilities currently unless somebody tries to integrate `IRremoteESP8266` into ESPHome as a dedicated component.

marius-arch, 2024-07-14:
Ok thank you