esphome/feature-requests #2787: Websocket transport option in MQTT client

grest, 2024-06-27:
**Describe the problem you have/What new integration you would like**

I've created the cloudflare tunnel for my local mqtt broker, so I can access it from internet. Unfortunately it works only through websocket transport. Can the MQTT client configuration be extendend with websocket transport option?

**Please describe your use case for this integration and alternatives you've tried:**

This will enable nicely the access to local mqtt broker from esphome devices running outside of local HA network (second flat, summer house etc).

**Additional context**

dula34, 2024-12-20:
Did you find a solution?

dougiteixeira, 2026-04-26:
I created a feature request related to this; here's the link if you'd like to support it: https://github.com/orgs/esphome/discussions/3635

Since this request wasn't tagged with the MQTT component label, I couldn't find it before.

dougiteixeira, 2026-11-11:
https://github.com/esphome/esphome/pull/16056 merged and included in version 2026.1 1.0 (currently in beta).