esphome/feature-requests #2788: Climate IR support for Ferroli split unit

seven7seven, 2024-06-27:
**Describe the problem you have/What new integration you would like**

Support for IR control for Ferroli Split Units, ex. Ferroli Reflex C 12 UI split-type AC units, using the Climate component.

**Please describe your use case for this integration and alternatives you've tried:**

Tried the following integrations:
- midea_ir
- zhlt01
- daikin_ir
- also tried manually capturing IR signals, in both RAW and some formats that looked consistent (ex. AEHA)

No luck with any,

**Additional context**

Here's some additional info on the unit and remote (I was not able to find more online about models, protocols etc.), neither generically nor esphome-related.
[image]
[image]

Thank you!

nagyrobi, 2024-06-27:
> Tried the following integrations:

Try also `zhjg01`. You need this, until it's merged and released:
```
external_components:
  - source: github://pr#6996
    components: 
      - heatpumpir
    refresh: 5min
```

seven7seven, 2024-06-27:
> Try also `zhjg01`. You need this, until it's merged and released:

Thanks, just ran it with:

```
external_components:
  - source: github://pr#6996
    components: 
      - heatpumpir
    refresh: 5min

climate:
  - platform: heatpumpir
    horizontal_default: auto
    vertical_default: auto
    max_temperature: 30
    min_temperature: 16
    protocol: zhjg01
    name: "Bedroom AC"
    receiver_id: recvr
    transmitter_id: trans
    sensor: temperature
```

The on-board receiver does register pronto signals being sent, but as the others I've tried the AC does not operate.

The remote's pronoto codes are also quite different from the ones sent by the climate remote.

I assume it's some sort of othger standard, but have not been able to identify it.

Thanks!

seven7seven, 2024-06-29:
Solved using:

```
climate:
  - platform: tcl112
    name: "Bedroom AC"
    transmitter_id: trans
```

Ferroli apparently uses the fuego protocol.

Procsiab, 2024-11-11:
Hello there, I found this issue before resorting to try and contribute back to the project and wanted to add a bit more to help the next peron who will land here: I too have an internal unit from Ferroli, but that's an Aster S, which uses the `Airton` protocol (as decoded by [IRremoteESP8266](https://github.com/crankyoldgit/IRremoteESP8266)).

I can also add that, based on claims from the technician who recently made some maintenance to my unit, newer Ferroli models like the Giada are manufactured by Midea and it's very likely they use their protocol too. Also, latest models have a USB optional accessory to control them via WiFi, and there is a compatible microcontroller board available from [SMLight](https://smlight.tech/product/slwf-01/) that should also work with those Ferroli newer units.

I wanted to add this information here because this was the first result I came across while researching for the protocol used by my unit, since it's the only feature request result for "Ferrroli".